verilator/verilator #2182: Improve Verilating speed for large designs with repetition

yTakatsukasa, 2020-03-01:
This is a related to https://github.com/verilator/verilator/issues/2140#issuecomment-578728611.
I'd like to collect information before touching code. Any advice is appreciated.

### Goal
I'd like to improve verilating speed and C++ compilation time for large designs, especially with repetition such as many core SoC.

### Approach
Skip per-scope optimization for large module which is repeated.

- modules can be marked by a pragma
- the marked modules will not be optimized per scope (per instance)
    - the modules will be scoped just once and optimized inside it.  `(A)`
    - no constant propagation across the module boundary in V3Gate so that the module is kept compatible among scopes
    - other instantiations of the module have dummy ASTScope
    - after descope,  the optimized result of `(A)` will be shared in all scopes

### Background
I found that passes after per-scope optimization consumes a lot of CPU time and memory.
It's quite natural that per-instance(per-scope) optimization needs such resources.

On the other hand, per-instance optimization for larger block such as processor is expensive.
It must be meaningful to optimize inside the processor module, but uniquifying the processor module for each instance seems less beneficial.

Disabling per-instance opt. for such large module will
- Decrease verilating speed and memory consumption ( more capability )
- Smaller C++ code since all instances of such module share the C++ implementation
- Less compilation time due to smaller C++ code
- Improve simulation speed by smaller code (improved instruction cache hit ratio)

### Concerns
- Ordering issues
- how should instance specific op be treated ? such as  $display("%m")
    - maybe unsupport such modules
- pitfalls I ovelrooked

### Statistics
Here is a statistics for my test design ( 128 instances of processor core https://github.com/lowRISC/ibex)
```
Performance Statistics:

  Stage, Elapsed time (sec), 001_cells              0.000000
  Stage, Elapsed time (sec), 002_linkparse          0.004940
  Stage, Elapsed time (sec), 003_linkdot            0.009097
  Stage, Elapsed time (sec), 004_linkresolve        0.001068
  Stage, Elapsed time (sec), 005_linklvalue         0.000379
  Stage, Elapsed time (sec), 006_link               0.000421
  Stage, Elapsed time (sec), 007_param              0.071113
  Stage, Elapsed time (sec), 008_paramlink          0.196825
  Stage, Elapsed time (sec), 009_deadModules        0.031841
  Stage, Elapsed time (sec), 010_width              0.051896
  Stage, Elapsed time (sec), 011_widthcommit        0.012004
  Stage, Elapsed time (sec), 012_const              0.009935
  Stage, Elapsed time (sec), 013_assertpre          0.015303
  Stage, Elapsed time (sec), 014_assert             0.005705
  Stage, Elapsed time (sec), 015_wraptop            0.000046
  Stage, Elapsed time (sec), 016_const              0.007756
  Stage, Elapsed time (sec), 017_split_var          0.004342
  Stage, Elapsed time (sec), 018_split_var          0.000204
  Stage, Elapsed time (sec), 019_dearray            0.004603
  Stage, Elapsed time (sec), 020_linkdot            0.315843
  Stage, Elapsed time (sec), 021_begin              0.063780
  Stage, Elapsed time (sec), 022_tristate           0.029940
  Stage, Elapsed time (sec), 023_unknown            0.018268
  Stage, Elapsed time (sec), 024_inline             0.134317
  Stage, Elapsed time (sec), 025_linkdot            0.290331
  Stage, Elapsed time (sec), 026_const              0.012163
  Stage, Elapsed time (sec), 027_deadDtypes         0.009144
  Stage, Elapsed time (sec), 028_inst               0.010735
  Stage, Elapsed time (sec), 029_const              0.006795
  Stage, Elapsed time (sec), 030_scope              1.033281  <== most passes before scope takes less than 100ms
[... 104 lines cut ...]
  Stage, Memory (MB), 054_genclk                   3224.660156
  Stage, Memory (MB), 055_clock                    3224.660156
  Stage, Memory (MB), 056_const                    3224.660156
  Stage, Memory (MB), 057_life                     3224.660156
  Stage, Memory (MB), 058_life_post                3224.660156
  Stage, Memory (MB), 059_const                    3224.660156
  Stage, Memory (MB), 060_deadAllScoped            3224.660156
  Stage, Memory (MB), 061_changed                  3224.660156
  Stage, Memory (MB), 062_descope                  3224.660156
  Stage, Memory (MB), 063_localize                 3224.660156
  Stage, Memory (MB), 064_combine                  3224.660156
  Stage, Memory (MB), 065_const                    3224.660156
  Stage, Memory (MB), 066_deadAll                  3224.660156
  Stage, Memory (MB), 067_clean                    3358.722656
  Stage, Memory (MB), 068_premit                   3363.492188
  Stage, Memory (MB), 069_expand                   3364.136719
  Stage, Memory (MB), 070_const_cpp                3364.136719
  Stage, Memory (MB), 071_subst                    3364.136719
  Stage, Memory (MB), 072_const_cpp                3364.136719
  Stage, Memory (MB), 073_deadAll                  3364.636719
  Stage, Memory (MB), 074_reloop                   3364.636719
  Stage, Memory (MB), 075_depth                    3364.636719
  Stage, Memory (MB), 076_cast                     3364.636719
  Stage, Memory (MB), 077_cuse                     3364.636719
```

wsnyder, 2020-03-02:
Improving large design handling would be very valuable.  A lot of the potential gain may be making the gcc time less (versus Verilation time).  BTW 22 seconds is pretty fast in the grand context, but I presume that was just a test case to show the point.

I'm not sure of the tradeoffs of what you suggest versus say making "protect-like" modules. This idea should be much faster in runtime, but likely significant effort.  Maybe the ideas can be combined, that is you mark pragmas that are the repeated module boundaries, and Verilator understands how to make the repeated sub-modules at those boundaries with separate sub-runs?

A problem I see are items like JTAG and similar chains that wind through the modules. The order of evaluation will need to be correct across the little pieces, and called in a different order per repetition, or it will run too slow.  I think you'll need to order the "meta-module" considering all submodule repetitions at once.

%m ties in with VPI/DPI, which must know the full design scope.  I would think you can maintain a AstScope as currently for the whole design hierarchy. A given scope would then know it's a repetition and point at another module.  Currently every class that composes a module already can have multiple instances with different names.

I'd suggest maybe handcrafting what the output C++ code would look like for a trivial design and we can discuss. Then propose the steps and algorithms.

@toddstrader

toddstrader, 2020-03-02:
> I'm not sure of the tradeoffs of what you suggest versus say making "protect-like" modules. This idea should be much faster in runtime, but likely significant effort. Maybe the ideas can be [...]

Related to this, is your biggest pain point the initial time it takes you to compile the entire design?  Or is it when you are iterating on a file in the edit-compile-test loop?  I'd also be glad to see improvements on either front, but the latter would impact my daily workflow more and I would expect it to have a lower floor on execution time since there's less (new) work to be done.

yTakatsukasa, 2020-03-03:
Thanks for the comments.
> BTW 22 seconds is pretty fast in the grand context, but I presume that was just a test case to show the point.

Right. The example is just an example that can be shown in public.
I saw much bigger design that could not be verilated due to huge memory consumption, but I don't remember the detailed numbers.
I am using protected-lib to handle the design. (Thank you! Todd)

JTAG example is good point. It seems I thought things too easy.
I realized I need to learn more about the ordering code.
I'm quite new to the backend of verilator.
After I study, I will write a C++ code that would be generated.

> Related to this, is your biggest pain point the initial time it takes you to compile the entire design? Or is it when you are iterating on a file in the edit-compile-test loop?

Well, I didn't separate them in my mind, but it is important to ask myself and check my usecase.
At this moment, I always verilate & g++ from scratch, I don't use ccache.
I am also interested in #1520.

yTakatsukasa, 2020-03-09:
After I explored the ordering code, I think protect-lib based approach is better because the feature is already working great and enhancement for the feature may be beneficial for more users.

I think the following items will help users with large design. (including me)

#### Usability for the usecase of veirlating speed
- hierarchical  verilation based of protect-lib can be done just adding some pragma to modules
    - simpler command line

### C++ compile time
- deterministic C++ code generation (more ccache friendly)
- skip emitting code if AST is same (check not only by timestamp)

### Simulation Performance
- #1520

yTakatsukasa, 2020-03-13:
I'd like to start from the usability, simple hierarchical verilation based on protect-lib feature.

Here is what I am thinking.

Assume the design looks like this. Note that sub_s, sub_b, and sub_c are marked `hier_block`.
```verilog
module sub_a() /*verilator hier_block*/;
endmodule

module sub_b() /*verilator hier_block*/;
   sub_a i_sub_a();
endmodule

module sub_c() /*verilator hier_block*/;
endmodule

module sub_d();
endmodule

module top();
   sub_b i_sub_b();
   sub_c i_sub_c();
   sub_d i_sub_d();
endmodule
```

Running verilator and make as below would generate an executable. There will be nothing special.
```bash
verilator --cc --exe --top-module top sub_a.sv sub_b.sv sub_c.sv sub_d.sv top.sv
make -f Vtop.mk -C obj_dir
```

Internally verilator would create protect lib for sub_a, sub_b, and sub_c to shorten verilation time.
There will be files as below.
```bash
obj_dir/Vtop.mk
obj_dir/Vtop_*.h
obj_dir/Vtop_*.cpp
obj_dir/Vsub_a/Vsub_a.mk
obj_dir/Vsub_a/Vsub_a_*.h
obj_dir/Vsub_a/Vsub_a_*.cpp
obj_dir/Vsub_b/Vsub_b.mk
obj_dir/Vsub_b/Vsub_b_*.h
obj_dir/Vsub_b/Vsub_b_*.cpp
obj_dir/Vsub_c/Vsub_c.mk
obj_dir/Vsub_c/Vsub_c_*.h
obj_dir/Vsub_c/Vsub_c_*.cpp
```

The same thing can be done with the current verilator, but user have to call verilator for each protect-libs and need to pass correct file list and library flags.

Any comment is appreciated including meta comment name :)

yTakatsukasa, 2020-10-04:
#2206 covers this issue.
https://github.com/verilator/verilator/issues/2578#issue-713273428 reports that `--hierarchical` helps verilate large designs though there is still demand to improve performance.

closed 2020-10-04 by yTakatsukasa