verilator/verilator #2184: Combinational logic in always_ff sensitivity list

dpetrisko, 2020-03-06:
```
module foo
 (input clk_i
  , input reset_i
  , input freeze_i
  );

  always_ff @(negedge (reset_i | freeze_i))
    $display("Reset sequence finished");

endmodule
```

Result: 
```
$ verilator --cc foo.v
%Error: foo.v:8: syntax error, unexpected '|', expecting ')'
%Error: Exiting due to 1 error(s)
%Error: Command Failed /usr/bin/verilator_bin --cc foo.v
```

```
module bar
 (input clk_i
  , input reset_i
  , input freeze_i
  );

  wire any_reset_li = reset_i | freeze_i;
  always_ff @(negedge any_reset_li)
    $display("Reset sequence finished");

endmodule
```

Passes fine, though.  foo.v works fine in Synopsys VCS

wsnyder, 2020-03-06:
Verilator doesn't support expressions as sensitivity lists, as this is rare and hard to support in its full form it will be a while before this is fixed.  It's on the list to fix, so closing this as a duplicate.

dpetrisko, 2020-03-06:
Sorry! Couldn’t find the duplicate!

wsnyder, 2020-03-07:
This is in our spreadsheet of work towards IEEE, but you're right there's no issue on it.

If you or someone wants to work on it, we would be glad to give guidance.

nakengelhardt, 2020-04-15:
Ran into a similar problem trying to get a coverage report for a verilator testbench - the coverage tool will try pinning the clock, which leads to the following error:
```
%Error: mutated.sv:152: syntax error, unexpected INTEGER NUMBER, expecting IDENTIFIER or '('
  always @(posedge 1'h1 or posedge _23_)
                   ^~~~
%Error: Exiting due to 1 error(s)
```
Would it be easier to add a fix for the special case of constant expressions? They just need to be ignored...

wsnyder, 2020-04-15:
Thought that was supported, guess not.  The internals I believe will already add a constant.

Can you attempt a pull request?  It would need a test_regress test (see internals.doc), then in verilog.y add yaINTNUM to event_expression.

wsnyder, 2023-01-29:
The version 5.000 scheduler accepts this example.

closed 2023-01-29 by wsnyder