verilator/verilator #2185: Speeding up the compilation with -jx

Dolu1990, 2020-03-08:
Hi,

I'm currently trying to speed up the compilation of the verilated C++ model from a large netlist.

I successfuly asked Verilator to split the model into many cpp files : 

```
 --output-split 5000
 --output-split-cfuncs 500
 --output-split-ctrace 500
``` 

But when i make the xxx.mk generated by verilator, all those verilated cpp files are compiled as a single cpp via the generated Vxxx__ALLcls.cpp : 

```
// DESCRIPTION: Generated by verilator_includer via makefile
#define VL_INCLUDE_OPT include
#include "Vunnamed.cpp"
#include "Vunnamed__1.cpp"
#include "Vunnamed__2.cpp"
#include "Vunnamed__3.cpp"
#include "Vunnamed__4.cpp"
#include "Vunnamed__5.cpp"
#include "Vunnamed__6.cpp"
#include "Vunnamed__7.cpp"
#include "Vunnamed__8.cpp"
#include "Vunnamed__9.cpp"
#include "Vunnamed__10.cpp"
#include "Vunnamed__11.cpp"
#include "Vunnamed__12.cpp"
#include "Vunnamed__13.cpp"
#include "Vunnamed__14.cpp"
#include "Vunnamed__15.cpp"
#include "Vunnamed__16.cpp"
#include "Vunnamed__17.cpp"
#include "Vunnamed__18.cpp"
#include "Vunnamed__19.cpp"
#include "Vunnamed__20.cpp"
#include "Vunnamed__21.cpp"
#include "Vunnamed__22.cpp"
#include "Vunnamed__23.cpp"
#include "Vunnamed__24.cpp"
#include "Vunnamed__25.cpp"
#include "Vunnamed__26.cpp"
#include "Vunnamed__27.cpp"
#include "Vunnamed__28.cpp"
#include "Vunnamed__29.cpp"
#include "Vunnamed__30.cpp"
#include "Vunnamed__31.cpp"
#include "Vunnamed__32.cpp"
#include "Vunnamed__33.cpp"
#include "Vunnamed__34.cpp"
#include "Vunnamed__35.cpp"
#include "Vunnamed__Slow.cpp"
#include "Vunnamed__1__Slow.cpp"
#include "Vunnamed__2__Slow.cpp"
#include "Vunnamed__3__Slow.cpp"
#include "Vunnamed__4__Slow.cpp"
#include "Vunnamed__5__Slow.cpp"
```

Is there a way to avoid that ? Likely compiling all those files separatly would take advantage of a make -j$(nproc)

Or i missed something ?

wsnyder, 2020-03-08:
You need to make with VM_PARALLEL_BUILDS=1.

closed 2020-03-08 by wsnyder

wsnyder, 2020-03-08:
Using VM_PARALLEL_BUILDS was documented but as was a source for errors, pushed to git towards version 4.032 to default this flag when using --output-split.

Dolu1990, 2020-03-08:
Nice thanks :D
VM_PARALLEL_BUILDS work quite well for me.