verilator/verilator #2187: Optimize replicate-and-or decoders

wsnyder, 2020-03-08:
Looking at a profile of Cores-SewRV, dec_tlu_ctl.sv line 2495 is the hotest block.

This implements a large decoder and REPLICATE-AND-OR mux like this:

   assign csr_misa = (!dec_csr_rdaddr_d[11]&!dec_csr_rdaddr_d[6]
       &!dec_csr_rdaddr_d[5]&!dec_csr_rdaddr_d[2]&dec_csr_rdaddr_d[0]);
   assign csr_mvendorid = (dec_csr_rdaddr_d[10]&!dec_csr_rdaddr_d[7]
       &!dec_csr_rdaddr_d[1]&dec_csr_rdaddr_d[0]);
   assign dec_csr_rddata_d[31:0] = ( ({32{csr_misa}}      & 32'h40001104) |
                                     ({32{csr_mvendorid}} & 32'h00000045) |

Optimize REPLICATE-AND-OR muxes to be case statements or otherwise (that then later
get optimized.)  As the decode provided is sparse this might be difficult, requires thought how to do this.  Perhaps all replicate-ands should be candidates.

See also #2186.

Committing t_gate_ormux.v to show example.

wsnyder, 2020-05-29:
@gezalore - FYI this is also an optimization that was suggested from analyzing EL1.

gezalore, 2025-06-16:
Assume this was meant for #6096. Let's keep the cancel-in-progress, I myself tend to push many times.