verilator/verilator #2188: How multiple assignments should be handled?

PetrGlad, 2020-03-08:
I use Verilator for linting as `verilator --lint-only -Wall myModule.v`. 
Linting following code does not produce any warnings (I am using yosys afterwards and it is happy to process it too)
```
reg some;
assign some = 1;
assign some = 0;
```
not only this is confusing, the effect of this code depends on assignment order. That is in the code above `some` is always 1, while in the code below it is always 0
```
reg some;
assign some = 0;
assign some = 1;
```
same for
```
reg some = 0;
assign some = 1;
```
the first assignment takes precedence in yosys. But in my case it was source of a hard to find bug because assignments were far away from each other. I found similar ticket (alhough not for continuous assignments) for that seem to be automatically closed #1939 (I think it should have been a feature request instead).

So the questions are: what is expected behavior here, should this be reported by linter, and if not what is recommended way to handle this?

I use 
```
verilator --version
Verilator 4.016 2019-06-16 rev UNKNOWN_REV
```
built from 18f8cd05 (master).

wsnyder, 2020-03-09:
It should give a warning on this, though note IEEE says it is optional to warn in section 6.5.

A quick look at this suggests it is not as simple as IEEE suggests, as multiple assignments to Z are legal (and the Z isn't known at the time of most warnings). Additionally each bit needs to be separately examined.

Maybe only look for assignments to non-Z constants....

PeterMonsson, 2020-09-16:
I would like to try handling this (for the non-z case). 

Where would it be best to handle this? All tips and tricks are greatly appreciated

wsnyder, 2020-09-16:
Great! I'd suggest in V3Undriven, which looks for undriven nets, this would be a double-driven case.  Please test cases with single bits multi-driven, and a whole bus driver with one bit driven, and likewise with assignments to Z.  I don't think you'll be able to fully ignore Z's or will be too many false positives, perhaps any expression with a Z underneath turns off checking for that variable.

sashimi-yzh, 2021-09-05:
I also need this feature to detect some design errors. Any plan to support this?

PeterMonsson, 2021-09-10:
I haven't had time to work on this for a while. Full time consulting and a newborn has erased any spare time. I am open to let others work on this. Maybe my schedule will have room in summer 2022, but I can't say for sure. 

It is also possible to buy my working hours if someone really needs this as I am an independent consultant. I can make sure that there is room in my schedule in the summer 2022 for paid work.

sifferman, 2022-12-03:
If possible, I'd like to bump this up. I've had students write code with this and it would be nice for Verilator to automatically catch it. This and #1184.

wsnyder, 2022-12-04:
This duplicates #1184

closed 2022-12-04 by wsnyder