verilator/verilator #2190: Support legacy file descriptors for $fgetc

thoughtpolice, 2020-03-13:
While using [Bluespec](https://github.com/b-lang-org/bsc) to generate my Verilog testbench, I found out that Verilator does not support legacy file descriptors (described by constant values) when calling `$fgetc`). My use case here is that my testbench reads data from `stdin` and writes it to `stdout` so I can `diff` it.

More concretely, this does not seem to work in the generated Verilog (`dut` is the harness containing the `$fgetc` call):

```
$ verilator ... --cc --exe --trace --top-module top testbench.cc tbtop.v dut.v keccak.v

%Error: dut.v:696: Internal: Unexpected Call
                 : ... In instance top.uut
   b__h12407 = $fgetc(32'h80000000);
               ^~~~~~
%Error: Internal Error: dut.v:696: ../V3AstNodes.h:4717: Unexpected Call
                                 : ... In instance top.uut
   b__h12407 = $fgetc(32'h80000000);
               ^~~~~~
```

It is *probably* possible to change the Bluespec compiler to fix this. However, there is some support for these constant legacy descriptors already, I think; for example, #961 describes this same basic problem, only with `$fwrite` instead.

I would be willing to write a patch for this, but the relevant code seems to have changed quite a bit since #961 was fixed, so I'm not sure where to start...

wsnyder, 2020-03-13:
Sure, would be good to have a patch in this area.  For details on process see the internals.adoc.  The process would be roughly:
1. Write a test_regress test of the MCD format calls, you can probably clone t_sys_file or something using the new format calls.  Make sure this works (and self-checks) on another simulator e.g. Icarus.  
2. Fix parser (verilog.y and whatever else breaks) to make the calls get through into C++.
3. Fix include/verilated.h VL_CVT_I_FP and whatever else to properly convert a MCD to a FILE*.

The hard part will be to support output to two streams in parallel (the "M" in MCD), perhaps punt on that?

wsnyder, 2020-05-14:
@stephenry thanks for fixing this.

closed 2020-05-14 by wsnyder

thoughtpolice, 2020-05-28:
That looks so much better than my poor attempts I tried. :) Thank you @stephenry! I'll give a try with Bluespec when I get a chance...