verilator/verilator #2191: Standard format for error messages

cpitclaudel, 2020-03-14:
Hi there,

Thanks a lot for the hard work on Verilator!  Flycheck (an Emacs plugin that runs linters as a file is being edited) has [support for Verilator](https://github.com/flycheck/flycheck/blob/08345d38e2872a3dcb14228edff796195809266f/flycheck.el#L11212), but our patterns are quite incomplete.  It seems some errors and warnings are prefixed with `%%` and other with a single `%`, and some warnings and errors but not all have an error id like `WIDTH` or `UNUSED` suffixed to them. Some errors like `%Error: No top level module found` have no line or column number, but not all of these should be reported to the user (e.g. `%Error: Exiting due to 2 warning(s)`).

Is the format of errors returned documented somewhere? Alternatively, it would be wonderful to have a standardized tool-friendly format (like JSON).

Thanks again for the hard work!

wsnyder, 2020-03-14:
There should be no %%'s.  I'll come up with a format regexp and make a test that makes sure the formats match that.

Some notes:
1. Use --quiet-exit to remove the "Command Failed' stuff.
2. I'm thinking it is useful to add column numbers ala clang, does this help?
3. Newest gcc's prefix all quoted lines with file line markers.  Do you think this useful?

wsnyder, 2020-03-14:
4. Perhaps --quiet-exit should suppress the "Exiting due to'?

wsnyder, 2020-03-15:
Committed following:
2. Added column numbers.
3. Not yet adding GCC lines but may in future, seems you're not parsing these anyways.
4. --quiet-exit removes "Exiting due to" in addition to previous "Command Failed". Please use this.
5. Tested and documented error format in manual:

             %(Error|Warning)(-[A-Z0-9_]+)?: ((\S+):(\d+):((\d+):)?)? .*

wsnyder, 2020-03-15:
Feel free to reopen if I missed something else.

closed 2020-03-15 by wsnyder

cpitclaudel, 2020-03-15:
Fabulous! Thanks a lot for the very quick turnaround (and for the hard work on this wonderful tool) :)
The addition of columns is great too.

Btw, we've recently added support for error ranges too, so if Verilator has that information (the end column of the error, in addition to the beginning), we could display it.

> Newest gcc's prefix all quoted lines with file line markers. Do you think this useful?

GCC's output has always been tricky to parse.  I haven't seen the latest change.  In general the very best is a format like JSON or XML that's unambiguous (eg. in ESLint https://eslint.org/docs/user-guide/formatters/ or Rust https://github.com/rust-lang/rust/blob/master/src/librustc_errors/json.rs or Checkstyle for a host of tools, eg. https://github.com/phpstan/phpstan/issues/271).  Otherwise it helps if the beginning of an error is unambiguous (for example if all lines except the first one are indented)

cpitclaudel, 2020-03-17:
Should there really two blanks in that regexp?

```
%(Error|Warning)(-[A-Z0-9_]+)?: ((\S+):(\d+):((\d+):)?)? .*
                               ^ here                   ^ and here
```

wsnyder, 2020-03-17:
Blank should be inside the ?

    %(Error|Warning)(-[A-Z0-9_]+)?: ((\S+):(\d+):((\d+):)? )?.*