verilator/verilator #2196: Option to Not Use ar while compiling

jshaker000, 2020-03-22:
I would like to have the option to run verilator and have its generated makefile NOT use ar to archeive the object files during compilation. I am running on macos and my compiler cannot read the archive files.  

IE:
   
`ar -cr Vdouble_dabble__ALL.a Vdouble_dabble__ALLcls.o Vdouble_dabble__ALLsup.o
ranlib Vdouble_dabble__ALL.a`
`clang++    double_dabble.o verilated.o verilated_vcd_c.o Vdouble_dabble__ALL.a    -o Vdouble_dabble -lm -lstdc++ `

Fails with 

`ld:  warning: ignoring file Vdouble_dabble__ALL.a, file was built for archive which is not the architecture being linked (x86_64): Vdouble_dabble__ALL.a`

But running

`clang++  double_dabble.o verilated.o verilated_vcd_c.o Vdouble_dabble__ALLcls.o Vdouble_dabble__ALLsup.o   -o Vdouble_dabble -lm -lstdc++` 

Works. Having a hook to have the generated makefile skip that step would be appreciated for me. I can attach the relevant code and commands, but I dont think it is too

wsnyder, 2020-03-23:
I'm not opposed to fixing this, but I believe others have not had problems with MacOs's ar, and this thread suggests it might just be a problem with your PATH, can you check that first?

https://stackoverflow.com/questions/22107616/static-library-built-for-archive-which-is-not-the-architecture-being-linked-x86

jshaker000, 2020-03-23:
Thank you, yes that was the problem. Brew installed the gnu binutils and put them first in my path.
It might be nice adding the flag anyway, but things do work for me now.

wsnyder, 2020-03-23:
Great!

closed 2020-03-23 by wsnyder