verilator/verilator #2197: Malformed enum declaration causes hang

stephenry, 2020-03-23:
Against: "Verilator 4.031 devel rev v4.030-14-gc2b49f01" (a recent development build).

A malformed enum declaration (circular reference) should raise an error but instead causes a hang. Below:

typedef enum logic { A  = 'b0, B = 'b1 } foo; // GOOD
typedef enum foo { C  = 'b0, D = 'b1 } foo; // BAD

wsnyder, 2020-03-23:
Makes sense, perhaps you'd be willing to make a test and fix for this (see docs/CONTRIBUTING.adoc), basically make a test_regress file, then in V3Width check for recursion and error out, there's similar checks for other types of such errors.

stephenry, 2020-03-23:
I will take a look. I am currently in transit, so please give me a couple of days.

wsnyder, 2020-05-27:
This was fixed with #2319, forgot to close this too.

closed 2020-05-27 by wsnyder