verilator/verilator #2198: VCD dumping no longer working in 4.030

juliusbaxter, 2020-03-24:
Something has changed the way VCD dumping is done between 4.028 and 4.030 to make our VCD dumping infrastructure stop working.

I suspect it's something in this commit which addressed #2126 : https://github.com/verilator/verilator/commit/30a33a6104cd559629fa5cc54d1c0690d2615f20

I can't quite put my finger on it though.

I notice in the documentation the `#ifdef VM_TRACE` lines have been removed from the docs, but I don't see how that could completely cease VCD dumping for us.

Have I missed a post or some documentation elsewhere which addresses this?

I'm debugging now and I'll probably come to the solution, but it's definitely the case that (on OS X where I'm building this daily) when I was on 4.028 (installed via homebrew) my VCD dumping was working and upon upgrading to 4.030 it stopped. If I roll back to the 4.028 the exact same toplevel C++ file works to dump VCDs.

wsnyder, 2020-03-24:
Sorry this broke.

There was no intent to break the historical tracing methods, so there's no documentation what needs to change :).

I just grabbed the 4.028 tracing example and ran it on the trunk and it did make waves. So it looks like whatever I broke isn't immediately obvious to me.  Can you please provide an example or ideally debug?  I would suspect there's something different about the order of calls you make to the tracing class versus the examples.

juliusbaxter, 2020-03-24:
No worries, and thanks for your response. I'm trying to make a testcase which reproduces this that I can submit. I should have an update for you within a day or so.

juliusbaxter, 2020-03-24:
I have a test case which shows this.

[verilator-issue-2198.tar.gz](https://github.com/verilator/verilator/files/4375346/verilator-issue-2198.tar.gz)

You should just be able to extract it and do `make all` and it'll show a 0 size VCD at the output on 4.030 and a 3.6kB VCD output on 4.028.

I don't quite understand yet why it's not working against 4.030 but I'm using a slightly unconventional way of creating the `tfp` trace object.

Comment out below is the conventional way which works fine. And is what I'm going to use going forward.

Our method comes from inheriting what is done on the `rocket-chip` project: https://github.com/chipsalliance/rocket-chip/blob/master/src/main/resources/csrc/emulator.cc#L261

Well, not quite the same, because we want to have a global `tfp` object and my C++ isn't good enough to know how to create the pointer using the syntax they are, so it's slightly modified.

Anyway, for those using rocket-chip it may cease to work, but otherwise switching to the method which is in the FAQ works fine for us. So I wouldn't spend too much time investigating.

juliusbaxter, 2020-03-24:
To elaborate on the above, where I say we have a slightly unconventional way of creating the tracing object, in the top of the C++ file we have:

```
// From rocket-chip verilator.h

class VerilatedVcdFILE : public VerilatedVcdFile {
 public:
  VerilatedVcdFILE(FILE* file) : file(file) {}
  ~VerilatedVcdFILE() {}
  bool open(const std::string& name) override {
    // file should already be open
    return file != NULL;
  }
  void close() override {
    // file should be closed elsewhere
  }
  ssize_t write(const char* bufp, ssize_t len) override {
    return fwrite(bufp, 1, len, file);
  }
 private:
  FILE* file;
};

VerilatedVcdC* tfp = NULL;
```

and then in our `main()` we have some code which sets up a file descriptor when passed the `-v` switch on the command line:

```
  FILE * vcdfile = NULL;

...

      case 'v': {
        vcdfile = strcmp(optarg, "-") == 0 ? stdout : fopen(optarg, "w");
        if (!vcdfile) {
          std::cerr << "Unable to open " << optarg << " for VCD write\n";
          return 1;
        }
        break;

...

```

Finally further on down in `main()` we have the creation of the `tfp` object:

```
  Verilated::traceEverOn(true); // Verilator must compute traced signals
  std::unique_ptr<VerilatedVcdFILE> vcdfd(new VerilatedVcdFILE(vcdfile));
  tfp = new VerilatedVcdC(vcdfd.get());
  if (vcdfile) {
    dut->trace(tfp, 99);  // Trace 99 levels of hierarchy
    tfp->open("");
  }
```

And that's what doesn't work.

Simply switching to:

```
  tfp = new VerilatedVcdC;
  dut->trace(tfp, 99);  // Trace 99 levels of hierarchy
  tfp->open("lfsr.vcd");  
```

Does work.

wsnyder, 2020-03-24:
Had added a check that filenames shouldn't be empty, but forgot open could be overloaded like your code was doing. Fixed towards 4.032.

closed 2020-03-24 by wsnyder

juliusbaxter, 2020-03-24:
Thanks Wilson!