verilator/verilator #2199: For unique case use：Unsupported: RHS of ==? or !=? must be constant to be synthesizable

hdzhangdoc, 2020-03-24:
Hello，we met a problem with systemverilog‘s "unique case“ statement，hope your help：
[image]

source code is like below：
[image]

Could you help to solve it？Thanks

wsnyder, 2020-03-24:
The base error is something correctly unsupported, but it looks like the optimizer is doing something wrong in this case to get that message.

I'll need a standalone test that reproduces it to debug further, ideally in test_regress format.  

P.S. Attaching at least the error message as text instead of a screenshot would be make this a bit easier, as can cut & paste.

hdzhangdoc, 2020-03-26:
I'm sorry for that I didn't provide some src code， please check it below which the error exists in. Does it can help？

hdzhangdoc, 2020-03-26:
And another similar error occur，source code attached

[image]

hdzhangdoc, 2020-03-26:
[dm_csrs.zip](https://github.com/verilator/verilator/files/4385524/dm_csrs.zip)

hdzhangdoc, 2020-03-26:
[riscv_decoder.zip](https://github.com/verilator/verilator/files/4385527/riscv_decoder.zip)

wsnyder, 2020-03-26:
By example, I mean please provide something standalone and smallish which when I type "make" gives the error.  What you attached are not complete.  Also please try to remove whatever code is not relevant (that is you still get the error after removing). Thanks

hdzhangdoc, 2020-03-26:
my project is very huge， it perhaps will spend long time to simplify the project..

hdzhangdoc, 2020-03-30:
[verilator_riscv_test.zip](https://github.com/verilator/verilator/files/4400818/verilator_riscv_test.zip)
Please help，thanks

wsnyder, 2020-03-30:
Thanks, was able to reduce the test case. The key thing to get the error was --assert.

Fixed in git towards 4.032.

closed 2020-03-30 by wsnyder