verilator/verilator #2203: Question: can verilator give me COi or a list loads/drivers ?

kaddkaka, 2020-03-25:
Is it possible to use verilator to get/dump the (combinational) cone of influence for a signal? Or some way to print all loads or drivers for a specific signal?

My aim in the end is to be able to see if there is any dependency between two signals A and B.

wsnyder, 2020-03-25:
If you run with debug, Verilator will create .dot files you can view (see docs/internals.adoc) which show connectivity.  The "gate" state is probably closest to what you want. Another alternative is the --cdc option which will assign clock domains which might be what you want this for.

I presume what you want is one-time use, none of these are intended for production as there's no real support nor stability on the debug files.

closed 2020-03-25 by wsnyder

kaddkaka, 2020-03-25:
Thanks I will try it out and see what I can get. Abtually I would have liked this stable and to use it for regression tests.

What I want to check is that a certain signals drives a specific cell and nothing else.

kaddkaka, 2020-03-26:
The dot file is a bit difficult to use because there are a lot of nodes that are only graphical and doesn't contain the name of any wire from the source code, I guess a script could easily filter them. I'm thinking about modifying the the verilator source code to be able to dump what I want in a more clear way. So far I have only cloned the repo and looked into `bin/verilator` any hints where I should start looking are highly appreciated.

wsnyder, 2020-03-26:
V3Cdc is probably a good stating point.