verilator/verilator #2204: Expose VPI cbNextDeadline

mballance, 2020-03-25:
https://github.com/verilator/verilator/blob/590b1853d00714a369abfd80e237a8440d04a37c/include/verilated_vpi.cpp#L424

Verilator's VPI implementation includes a cbNextDeadline() method that returns the time of the next VPI callback. Currently this method is neither exposed via a header file, nor used by the implementation. I believe this API would be quite useful to integrations (ie C++ testbenches) that need to support VPI (thinking of cocotb, specifically). Any objections to exposing this in verilated_vpi.h? If not, I'll put in a PR.

Thanks,
Matthew

wsnyder, 2020-03-25:
Sure, please make a pull request for this & please add or improve a test to appease code coverage.

wsnyder, 2020-03-30:
Closed with pull #2212.

closed 2020-03-30 by wsnyder