verilator/verilator #2205: ” Duplicate declaration of TYPEDEF“ reported error，when typedef in a for statement

hdzhangdoc, 2020-03-26:
If I moved the typedef statement out of the for loop，the error diappeared，is it a bug？The source code can be compiled pass under VCS.

[image]

[image]

314   for (genvar t = 0; t < AxiMaxReads; t++) begin: gen_read_upsizer
315     r_state_t r_state_d;
316     r_state_t r_state_q;
317
318     // Are we idle?
319     assign idle_read_upsizer[t] = (r_state_q == R_IDLE);
320
321     struct packed {
322       ar_chan_t ar         ;
323       logic ar_valid       ;
324       logic ar_throw_error ;
325       slv_r_chan_t r       ;
326       logic r_valid        ;
327       burst_len_t burst_len;
328       size_t orig_ar_size  ;
329     } r_req_d, r_req_q;

wsnyder, 2020-03-26:
Confirmed.

wsnyder, 2020-03-26:
Thanks for your report.   Fixed in git towards 4.032.

closed 2020-03-26 by wsnyder