verilator/verilator #2208: Accessing elements of SystemVerilog struct from C++ test bench

mattofak, 2020-03-26:
I'm simulating a module that uses packed structs in its interface and attempting to test that module in C++. E.g.:

```systemverilog
typedef packed struct {
  logic a;
  logic b;
} foo_t;

module bar(
   foo_t foo;
)
endmodule
```

I was initially expecting to be able to be able to do something like
```c++
Vbar uut{};
uut.foo.a = 1;
```

But it appears that Verilator only defines the interface in terms of vectors. I couldn't find any reference to mask/match/shift information or any struct I could cast the vector to in order to make accessing the fields in the struct easier.

Am I missing something? Is there a best practice for this kind of interface?

If all else fails, I'll create my own representation of the fields and cast back/forth, but that feels super ugly and error prone, mostly because it's not connected to the verilog header and could get out of sync.

stephenry, 2020-03-26:
The standard approach is to use an unpacked struct, which (I believe) typically has the same memory layout as that of the underlying machine (though only for SV types that have a corresponding representation in C). I don't believe Verilator supports unpacked structures at the moment, so what I typically do is unpack the fields manually and pass these as independent arguments through a DPI call and reconstruct as necessary in the TB. This works well. I've created pack/unpack routines myself in the past however unless this is done procedurally (to either parse the SV or to generate the C++/SV representations), this can be a bit tricky and for non-trivial structures definitions, discrepancies can creep-in between the RTL and C++.

wsnyder, 2020-03-26:
There isn't currently any information created on structs, a request for this is issue #860 

Note you can manually create bitfields in C to at least make it read easier, e.g.

    struct foo_t {
     union {
         uint packed:32;
         struct {
            int a:1;
            int b:1;

If you do this note the code won't be portable unless you do some ifdefs, as the endian rules vary.

wsnyder, 2020-03-26:
BTW of course contributions to #860 would be welcome, feel free to comment there if you're interested in helping.

closed 2020-03-26 by wsnyder