verilator/verilator #2209: What is the intended/recommended way to view verilator dot files graphically

kaddkaka, 2020-03-27:
Hi, I've been fiddling with verilator, dot and ghostscript to view fotfiles outputted with `verilator --debug`. But my end result is always so bad resolution that it is unreadable. This is how I've done it:

```
verilator --cc --debug
dot -Tps2 -O ./obj_dir/Vverilator_048_gate_opt.dot
ps2pdf -dDEVICEWIDTHPOINTS=27389 -dDEVICEHEIGHTPOINTS=1903 -dEPSCrop ./obj_dir/Vverilator_048_gate_opt.dot.ps2
firefox Vverilator_048_gate_opt.dot.ps2.pdf
```

The first trick with p2pdf was to figure out that I needed specific DEVICEWIDTHPOINTS, otherwise The generated PDF was just standard format and I coul only view 1 % of my graph (it looked like the top left corner).

When I view it with firefox, I can soom in, but at 100 % zoom, everything is extremely blurry.

I also tried look at the generated ps file with ghostscript (once again needing -dDEVICEWIDTHPOINTS etc to not crop the image of the graph).
`ghostscript -dDEVICEWIDTHPOINTS=27389 -dDEVICEHEIGHTPOINTS=1903 ./obj_dir/Vverilator_048_gate_opt.dot.ps2`
This starts the ghostscript application with a really small windows, without any menu bar and I have no idea how to zoom or do anything except looking at this miniature. (I have never used ghostscript before so maybe there is some way to look at a portion of the graph with this tool)

How can I look at my graph with reasonable resolution? I'm sure there is some better way to do this that I haven't figured out. All help appreciated. :)

toddstrader, 2020-03-27:
I always just run the command that Verilator suggests when it creates the dot file.  For example:
```
dot -Tpdf -o ~/a.pdf obj_vlt/t_type_comparison/Vt_type_comparison_063_orderg_done.dot
```

This goes straight to PDF and is perfectly readable for me.

wsnyder, 2020-03-27:
ZGRViewer is good, but I've never found a great tool.  Look for any viewer that supports Graphviz format.  If you find anything good or have other notes please consider a pull request that adds suggestions to the .dot output section of docs/internals.adoc.

closed 2020-03-27 by wsnyder

kaddkaka, 2020-03-27:
@Todd Strader I tried that way also and the resolution for the PDF is the
same, unreadable

[image: image.png]

kaddkaka, 2020-03-27:
@toddstrader How big is your bounding box when you use that command? It might be that the bad resolution comes from not shear size of the pdf. My result looks like this: (the visible text is the biggest text in the graph)

[image]

kaddkaka, 2022-06-07:
Just checking in: Has anyone found a better way to view these files?

toddstrader, 2022-06-07:
@kaddkaka yikes, sorry I missed your question a while back.  I'm not sure how to check the bounding box size.  Evince tells me the PDF is 9.93 × 11.31 inch for whatever that's worth.  Here's a copy of what I'm looking at (same command, different dot file):
[a.pdf](https://github.com/verilator/verilator/files/8856537/a.pdf)

gezalore, 2022-06-07:
@kaddkaka `xdot` works quite well for me, at least for graphs that are reasonably sized.