verilator/verilator #2210: -incdir isn't supported for 4.030

agrobman, 2020-03-27:
Why?

Our setups, worked with 4.020, do not compile now! Please, don't remove/change command switch options in future, we have enough troubles with our design itself yet to deal with the tool updates!

wsnyder, 2020-03-27:
Sorry you are having problems.  In general, our intent is never to remove non-experimental features, except a few noted in the DEPRECATIONS section where we give a year's notice.

Specifically, incdir is still supported/intended to be supported.  I did a diff and there are no obvious diffs that would cause this problem, and there are incdir tests in the regression that pass.  There was a change to how quoted arguments are handled to make it more compatible with other simulators, maybe you are hitting that.  If you have an example that fails unexpectedly please attach a test case.

agrobman, 2020-03-27:
Hi Wilson,

You can try our Swerv  repos with new verilator …

flist files have –incdir options

Alex

Sorry you are having problems. In general, our intent is never to remove non-experimental features, except a few noted in the DEPRECATIONS section where we give a year's notice.

Specifically, incdir is still supported/intended to be supported. I did a diff and there are no obvious diffs that would cause this problem, and there are incdir tests in the regression that pass. There was a change to how quoted arguments are handled to make it more compatible with other simulators, maybe you are hitting that. If you have an example that fails unexpectedly please attach a test case.

toddstrader, 2020-03-27:
@agrobman please note that this CI environment was made specifically to test for regressions against downstream projects including SweRV:
https://travis-ci.com/github/verilator/verilator_ext_tests

It's currently green, so whatever it is doing in the SweRV repo is happy right now with HEAD of Verilator.  Can you take a look to see if this will reproduce the -incidr options you're having problems with?
https://github.com/verilator/verilator_ext_tests/blob/master/t/t_cores_swerv_cmark.pl

Specifically, it is doing something like this in the SweRV repo:
```
make -f tools/Makefile verilator snapshot=mybuild
```

agrobman, 2020-03-27:
Todd, Winson,

The SweRV command you are asking about won’t compile executable if it was compiled before.

Better to use:

make -f tools/Makefile

( which cleans work directory and builds executable )

Here is what I  get with my flist file starting with:

“

+libext+.v+.sv

-incdir $RV_ROOT/design/include

-incdir $SYNOPSYS_SYN_ROOT/dw/sim_ver

-y $SYNOPSYS_SYN_ROOT/dw/sim_ver

$RV_ROOT/design/elx2s_swerv_wrapper.sv

“

verilator '-UASSERT_ON' --cc -CFLAGS "-std=c++11" snapshots/default/common_defines.vh /wdc/proj/riscv/mustang/users/alexander.grobman/swerv_elx2s/design/include/elx2s_def.sv snapshots/default/elx2s_pdef.vh \

        -Isnapshots/default -I/wdc/proj/riscv/mustang/users/alexander.grobman/swerv_elx2s/testbench -f /wdc/proj/riscv/mustang/users/alexander.grobman/swerv_elx2s/testbench/flist \

        -Wno-WIDTH -Wno-UNOPTFLAT /wdc/proj/riscv/mustang/users/alexander.grobman/swerv_elx2s/testbench/tb_top.sv /wdc/proj/riscv/mustang/users/alexander.grobman/swerv_elx2s/testbench/ahb_sif.sv --top-module tb_top \

        -exe test_tb_top.cpp --trace --autoflush

%Error: Invalid Option: -incdir

make: *** [verilator-build] Error 1

the same design compiles if I replace –incdir with +incdir+,

Alex

@agrobman<https://github.com/agrobman> please note that this CI environment was made specifically to test for regressions against downstream projects including SweRV:
https://travis-ci.com/github/verilator/verilator_ext_tests

It's currently green, so whatever it is doing in the SweRV repo is happy right now with HEAD of Verilator. Can you take a look to see if this will reproduce the -incidr options you're having problems with?
https://github.com/verilator/verilator_ext_tests/blob/master/t/t_cores_swerv_cmark.pl

Specifically, it is doing something like this in the SweRV repo:

make -f tools/Makefile verilator snapshot=mybuild

agrobman, 2020-03-27:
sorry for misleading - Swerv EH1 doesn't have -incdir statements in flist file ...

wsnyder, 2020-03-27:
Note the Travis regression always builds from a clean tree, it never does incremental builds.

Verilator has never supported -incdir (vs +incdir+), as AFAIK the plus version is the standard one in other tools.  I can't explain how it ever worked as that parsing hasn't changed.

I presume you are referring to EH2. When the EH2 core is public we will add it to our regressions, but can't/won't until it is public. When I run this command on EH2 it passes.  The only use I see of -incdir is in a irun-build which is obviously a different tool.

I think something changed in your makefiles unrelated to Verilator and this isn't a Verilator bug.  Please confirm or provide a testcase that works on 4.020.

agrobman, 2020-03-27:
Wilson,

these -incdir statements were in our internal DBs from day 1 of using verilator 3.926 yet, and I think should be in rev1 of EH1. I inherited them from another designer, who set up the first demo TB.
Later I removed them from flist files as not needed. 

Anyway, previous post shows that verilator doesn't like them , although 4.020 was OK with them ..

wsnyder, 2020-03-27:
Please send a test case that works on 4.020 and fails on 4.030.

wsnyder, 2020-03-27:
(Note I'm not trying to be a pain, I'm sorry you're having this problem, but when I look at the sources it isn't obvious what could have changed this, so a test is important.)

agrobman, 2020-03-27:
Sorry, it was probably my fault: I added these for our new project to compile with xcelium

closed 2020-03-27 by agrobman

wsnyder, 2020-03-27:
Great!