verilator/verilator #2211: Verilator for Single-Event Effect fault injection

DinghyDog, 2020-03-27:
I would like to use Verilator to perform Single-Event Effect (SEE) fault injection, such as Single-Event Upsets in flip-flops and Single-Event Transients in gates*, to determine a failure rate on a functional level.

##### Background
Algorithmic or probabilistic fault analysis techniques have the problem that they analyze a static design and only calculate the fault propagation. In order to take the functional level of a design into account one has to perform fault injections during a circuit simulation. The first principle method to achieve this is to simulate the design (RTL or GLN) with a testbench representing a realistic workload and then randomly chose a target instance and time to inject the fault. The simulation is continued and it is evaluated if a failures can be observed appear on the functional level. However, this method can become very expensive in terms of computing resources and tool licenses. 

I was amazed by the speed of Verilator and was hoping I could develop a fault injection procedure for it. The additional fact that it is open source and free, thus does not require any licenses, makes it also very attractive to parallelize the fault injection simulations on a large scale (e.g. one parallel simulation per target flop-flop and injection time).

I started with a simple purely combinatorial circuit, the c17 from the ISCAS85 benchmark suite, and tried to inject Single-Event Transients. The SET model is simplified by just inverting the output of the target gate during the injection time/cycle. I verilated the module and build a small testbench in C++ iterating over all possible input values. For each input combination I invert the value of the target gate output and compare the module output against a fault-free reference.

In my first attempt I was able to modify the circuit and add a saboteur which modifies the target gate's output net. The net is simply inverted, depending if the fault injection is enabled or not 
`xor XOR(trg_net_fault, trg_net, fi_en)`
Although, this would be suitable to do for every net in this small circuit and maybe even scriptable for bigger circuits (such as other ISCAS85 benchmarks), I imagine it is getting quite complex for big designs with several hierarchies. This is why to my opinion, the ideal implementation would be build within the testbench, keeping the circuit untouched.

When I tried to modify the target net from the testbench, I noticed the difference between event driven and cycle based simulators (at least that is my understanding). Although the modified net seems to be taken over it does not propagate to the output. When I call the evaluation function of the module again, I guess the modified net gets overwritten by the previous drivers. I tried similar thing with VPI and the vpi_put_value function, but it seems to be the same problem.

Do you have an idea how I could achieve the described fault injection behavior and could point me in the right direction?

* I am aware that Verilator is not able to perform timing aware simulation and transients with different pulse length could not be simulated. However, the transient pulse can still be modeled in more abstract way and the propagation and effect on a functional level can still be analyzed. This analysis could be paired with a lower level electrical and timing analysis.

[verilator_fi.tar.gz](https://github.com/verilator/verilator/files/4395458/verilator_fi.tar.gz)

wsnyder, 2020-03-27:
## General comments:

At the gate level to get accurate results you really would want a gate level representation. As you probably know the SEU effects at flops are much more (10x+) important than transient or clock events which are often ignored.

If you will do a lot of simulations in this area note it might be more efficient to use formal techniques, or higher level architectural models, it only to first prove if a cone of logic is susceptible before simulation.

BTW you may be interested in the reference on part of this topic by my coworker, https://www.amazon.com/Architecture-Design-Errors-Shubu-Mukherjee/dp/0123695295

## On Verilator

If you are interested only in effects at flops, then Verilator could do this.  Verilator internally already knows what signals come from a posedge flop block.  You have two choices, 1. edit the resulting Ast tree to look at a target signal and vector, or 2. use the VPI or other methods to discover the storage behind the node and wiggle it in C++ code.  The second is probably an easier starting point. To begin you could mark everything --public and walk through the VPI tree and modify appropriately.

To make this higher speed instead of using --public, add a new option that when enabled in e.g. V3Delayed that marks every lvalue() signal under an AstAssignDly as public readable/writable.  By checking it's a delayed assignment you'll avoid changing combo logic and get most all flops.

DinghyDog, 2020-03-30:
Thank you very much for you answer and your helpful suggestions.

You are completely right, the accuracy at the gate level is generally higher (especially for SETs). To my understanding Verilator should be capable of using the gate level netlists, right? At least if the library primitives would be replaced with synthesizable Verilog descriptions. I am actually trying to do an assessment at the moment, and want to look at the difference in accuracy when injecting in the GLN and RTL. Also if there is a difference when using Verilator or an event driven simulator and the performance gain you get with Verilator.

Just to clarify this for me: in Verilator it is not possible to force a value of a wire (or reg), similar to the force or deposit command in Verilog? 

As you suggested I tried to modify the value of a flip-flop. The flip-flop is described as follows and  the SEU is injected into the Q_reg signal:
```
module dff (CK,Q,D);
input CK,D;
output Q;

  reg  Q_reg = 1'b0;

  always @ (posedge CK)
  begin
        Q_reg <= D;
  end

  assign Q = Q_reg;

endmodule
```
Instead of marking everything `--public` I used the `--public-flat-rw` Verilator flag which should do the same, right? I tried to modify the value directly via `dut->s27__DOT__DFF_0__DOT__Q_reg` or via the VPI function `vpi_put_value()`. After the injection I do a `dut->eval` (or `dut->eval_step();` and `dut->eval_end_step();`. I can see that the value of `Q_reg` has changed, but it does not propagate to the `Q` output or any other combinatorial logic. Do you have an idea why this is not working?

wsnyder, 2020-03-30:
>To my understanding Verilator should be capable of using the gate level netlists, right?

Verilator can simulate a gate level netist, though it may optimize the gates so aggressively and into such a different form as to be unsuitable to represent gate-level faults.  Many of these optimizations cannot be disabled.

>In Verilator it is not possible to force a value of a wire (or reg), similar to the force or deposit command in Verilog?

It is not possible to do this.

>Instead of marking everything `--public` I used the `--public-flat-rw` Verilator flag which should do the same, right?

Yes, for your purposes public-flat-rw is better, the difference between them is the C++ class scoping.

>I tried to modify the value directly via

Outputs coming only from a flop are only evaluated when that flop evaluates. So the value will not propagate until something needs it, e.g. it is clocked. You'd need this:

    always @ (posedge CK) begin
      if (sometimes) Q_reg <= D;
    end

Where "sometimes" is only sometimes set, then you clock it.

Alternatively though slower code something like this:

    input [31:0] fault;
    always @ (posedge CK) begin
      Q_reg_pre_fault <= D;
    end
    Q_reg = (fault == 1) ^ Q_reg;
    assign Q = Q_reg;

Then changing "fault" and calling eval() will cause an immediate change to Q.  Of course you could improve Verilator to make this transformation for you (if you do this there's probably a more efficient way as internally Verilator already has a previous and next version of each signal).

wsnyder, 2020-05-07:
I'm assuming you've gotten the feedback you needed on this discussion, if not feel free to open.

closed 2020-05-07 by wsnyder

TobiasFaller, 2021-04-30:
JFYI. A sample implementation for fault injection is available in PR #2724.