verilator/verilator #2213: MSVC: error C2010: '.': unexpected in macro parameter list

xobs, 2020-03-29:
Using MSVC 2019, I'm unable to use Verilator because it looks like MSVC isn't accepting some of the macros that are used.

For example, in `verilated_cov.h` `VL_COVER_INSERT` is defined as `#define VL_COVER_INSERT(countp,args...)`.  Is there some reason this macro is missing the comma between `args` and `...`?

```patch
--- a/include/verilated_cov.h
+++ b/include/verilated_cov.h
@@ -62,7 +62,7 @@
 ///             VL_COVER_INSERT(&m_cases[i], "comment", "Coverage Case", "i", cvtToNumStr(i));
 ///     }

-#define VL_COVER_INSERT(countp,args...) \
+#define VL_COVER_INSERT(countp,args,...) \
     VL_IF_COVER(VerilatedCov::_inserti(countp); \
                 VerilatedCov::_insertf(__FILE__, __LINE__); \
                 VerilatedCov::_insertp("hier", name(), args))
diff --git a/include/verilated_dpi.cpp b/include/verilated_dpi.cpp
index d25339de..9f4f9468 100644
--- a/include/verilated_dpi.cpp
+++ b/include/verilated_dpi.cpp
@@ -42,7 +42,7 @@
     VL_FATAL_MT(__FILE__, __LINE__, "", \
                 (std::string("%%Error: Unsupported DPI function: ")+VL_FUNC).c_str())

-#define _VL_SVDPI_WARN(message...) \
+#define _VL_SVDPI_WARN(message,...) \
     VL_PRINTF_MT(message)

 // Function requires a "context" in the import declaration
```

xobs, 2020-03-29:
Note that I'm still unable to use MSVC.  I'm getting a NULL pointer exception where `valps[4]` is NULL:

```
   VerilatedCovImp::insertp  lineno = 37
   VerilatedCovImp::insertp  page = sp_user/D:\Code\verilated-rs\target\debug\build\example-fa47c2cdb4c7b1fe\out\Vtop
   VerilatedCovImp::insertp  hier = TOP
Exception thrown at 0x00007FFB796A4111 (ucrtbase.dll) in example.exe: 0xC0000005: Access violation reading location 0x0000000000000000.
```

So maybe there is a semantic difference between `args...)` and `args, ...)`.

xobs, 2020-03-29:
Ah, there is a difference.  This patch fixes the compile error _and actually works_:

```patch
diff --git a/include/verilated_cov.h b/include/verilated_cov.h
index c0786ab1..3b8c2263 100644
--- a/include/verilated_cov.h
+++ b/include/verilated_cov.h
@@ -62,10 +62,10 @@
 ///             VL_COVER_INSERT(&m_cases[i], "comment", "Coverage Case", "i", cvtToNumStr(i));
 ///     }

-#define VL_COVER_INSERT(countp,args...) \
+#define VL_COVER_INSERT(countp,...) \
     VL_IF_COVER(VerilatedCov::_inserti(countp); \
                 VerilatedCov::_insertf(__FILE__, __LINE__); \
-                VerilatedCov::_insertp("hier", name(), args))
+                VerilatedCov::_insertp("hier", name(), __VA_ARGS__))

 //=============================================================================
 /// Convert VL_COVER_INSERT value arguments to strings
diff --git a/include/verilated_dpi.cpp b/include/verilated_dpi.cpp
index d25339de..7b4678ea 100644
--- a/include/verilated_dpi.cpp
+++ b/include/verilated_dpi.cpp
@@ -42,8 +42,8 @@
     VL_FATAL_MT(__FILE__, __LINE__, "", \
                 (std::string("%%Error: Unsupported DPI function: ")+VL_FUNC).c_str())

-#define _VL_SVDPI_WARN(message...) \
-    VL_PRINTF_MT(message)
+#define _VL_SVDPI_WARN(...) \
+    VL_PRINTF_MT(__VA_ARGS__)

 // Function requires a "context" in the import declaration
 #define _VL_SVDPI_CONTEXT_WARN() \
```

wsnyder, 2020-03-29:
Thanks for looking at this and making a fix.  The old code style is a holdover from before variable arg macros were standard.

If you don't mind can you please make it a pull request? Also please add your name to docs/CONTRIBUTORS matching your .gitconfig name.  This way it will get the regressions run against it automatically, and once passes I will merge it.  Thanks.

wsnyder, 2020-03-29:
Thanks, all merged.

closed 2020-03-29 by wsnyder