verilator/verilator #2216: supported 4-state numbers

hdzhangdoc, 2020-04-01:
Hello, When compiling systemverilog, I met an error report:
[image]
src code is below:
[image]
could we support it？

wsnyder, 2020-04-01:
This should work, please post a minimum self-runable test case showing this as text (not graphics), thanks.

wsnyder, 2020-04-05:
Doubt it's this, but found and fixed a case where converting to real numbers would give this error; pull from git to see if that was it.

wsnyder, 2020-05-01:
If you have a test case please post it, otherwise closing as need test.

closed 2020-05-01 by wsnyder

hdzhangdoc, 2020-06-03:
> Doubt it's this, but found and fixed a case where converting to real numbers would give this error; pull from git to see if that was it.

Hello，this probelm appears again under version 0529 master， could you help
to solve it？

wsnyder, 2020-06-03:
Please post a test case, thanks.

wsnyder, 2020-06-03:
Please post a test case that is self contained, that is code and a command line that gives the exact error, a code snip isn't enough, thanks.