verilator/verilator #2218: module default parameter didn't take effect

hdzhangdoc, 2020-04-02:
My code is shown below：
  27 module pulp_cluster
  28 #(
  29   // cluster parameters
  30   parameter bit    ASYNC_INTF              = 1'b1,
  31   parameter int    NB_CORES                = 8,
  32   parameter int    NB_HWPE_PORTS           = 0,
  33   parameter int    NB_DMAS                 = 4,
  34   parameter int    NB_EXT2MEM              = 2,
  35   parameter bit    CLUSTER_ALIAS           = 1'b1,
  36   parameter int    CLUSTER_ALIAS_BASE      = 12'h1B0,
  37   parameter int    TCDM_SIZE               = 320*1024,                 // [B], must be 2**N
  38   parameter int    NB_TCDM_BANKS           = 16,                      // must be 2**N
  39   parameter int    TCDM_BANK_SIZE          = TCDM_SIZE/NB_TCDM_BANKS, // [B]
  40   parameter int    TCDM_NUM_ROWS           = TCDM_BANK_SIZE/4,        // [words]
  41   parameter bit    HWPE_PRESENT            = 1'b0,                    // set to 1 if HW Processing Engines are present in the cluster
  42   parameter bit    SHARED_FPU              = 1'b1,
  43
  44   // I$ parameters
  45   parameter int    SET_ASSOCIATIVE         = 4,
  46   parameter bit    PRIV_ICACHE             = 1'b1,
  47   parameter bit    MP_ICACHE               = 1'b0,
  48   parameter bit    SP_ICACHE               = 1'b0,
  49   parameter int    NB_CACHE_BANKS          = PRIV_ICACHE ? 2 : 8,

。。。。。。

1029   if (PRIV_ICACHE) begin : gen_priv_icache
1030     icache_hier_top #(
1031       .FETCH_ADDR_WIDTH       ( 32                  ),
1032       .FETCH_DATA_WIDTH       ( 128                 ),
1033       .NB_CORES               ( NB_CORES            ),
1034       .SH_NB_BANKS            ( NB_CACHE_BANKS      ),
1035       .SH_NB_WAYS             ( 4                   ),
1036       .SH_CACHE_SIZE          ( 8*1024              ), // in Byte
1037       .SH_CACHE_LINE          ( 1                   ), // in word of [FETCH_DATA_WIDTH]
1038       .PRI_NB_WAYS            ( 4                   ),
1039       .PRI_CACHE_SIZE         ( 1 * 1024            ), // in Byte
1040       .PRI_CACHE_LINE         ( 1                   ), // in word of [FETCH_DATA_WIDTH]

I found that is didn't go into the ’if' branch，but go into the ‘else' branch which trigger the code error in ‘else’ branch。 the module initiats this module not give the PRIV_ICACHE one value。

wsnyder, 2020-04-02:
This should generally work correctly, please attach a standalone minimized test case, thanks.

wsnyder, 2020-05-01:
If you have a test please post it, otherwise closing for lack of test.

closed 2020-05-01 by wsnyder