verilator/verilator #2225: Function body with multiple statements should not pass linter in 1364-2001 language setting

laforest, 2020-04-04:
The following module passes linting as Verilog-2001 code despite having multiple statement in a function, which is a SystemVerilog (or Verilog-2005) feature. This error was detected by Icarus Verilog.

Linter script:
`verilator -Wall +1364-2001ext+v -y . --lint-only ${FILE}`

Module (with begin/end fix commented out):
```
module Gray_to_Binary_Reflected
#(
    parameter WORD_WIDTH = 32
)
(
    input  wire [WORD_WIDTH-1:0] gray_in,
    output reg  [WORD_WIDTH-1:0] binary_out
);

    localparam ZERO = {WORD_WIDTH{1'b0}};

    initial begin
        binary_out = ZERO;
    end

    function [WORD_WIDTH-1:0] gray_to_binary
    (
        input [WORD_WIDTH-1:0] gray
    );
        integer i;
        reg [WORD_WIDTH-1:0] binary;

//        begin
            binary[WORD_WIDTH-1] = gray[WORD_WIDTH-1];

            for(i=WORD_WIDTH-2; i >= 0; i=i-1) begin
                binary[i] = binary[i+1] ^ gray[i];
            end

            gray_to_binary = binary;
//        end
    endfunction

    always@(*) begin
        binary_out = gray_to_binary(gray_in);
    end

endmodule
```

wsnyder, 2020-04-04:
Verilator at present has very limited checking for "degraded" language settings. For the most part only lexing plus a few other things are language setting aware.

A patch would be appreciated for this, otherwise likely will be stuck behind higher-priority language features and linting of the most recent standard.

laforest, 2020-04-05:
Thanks for the explanation. I don't have the time right now to dive into a new codebase, and I agree that it's a minor problem not as important as newer language support. Please feel free to just close this issue.

(Aside: thanks again for Verilator. I use it daily for work, and it helps me get the code right the first time.)

wsnyder, 2020-04-05:
Thanks for filing it anyways and for the praise. We'll let this one rest ;)

closed 2020-04-05 by wsnyder