verilator/verilator #2238: Unsupported SystemVerilog when inferring BRAM on FPGA

oharboe, 2020-04-06:
I've written a bit of Verilog to infer BRAMs in an FPGA, but I'm having problems simulating this in Verilator.

It appears that this is not supported at this time, but I want to make sure that I'm not making some silly mistake:

```
**DualPortMaskedWrite.v:34: Unsupported: Delayed assignment to array inside for loops (non-delayed is ok - see docs)
                                                                                                                                                                                        
       ram[waddr][i] <= wdata[i * 8 +: 8];
                 ^
%Error: Exiting due to 1 error(s)
        ... See the manual and https://verilator.org for more assistance.
```

```
/**
 * From the Quartus Prime Standard Edition Handbook Volume 1
 * 
 * Based on "Example 75. SystemVerilog Simple Dual-Port Synchronous RAM with Byte Enable"
 */

module DualPortMaskedWrite(
	we, clk,
	waddr, raddr,
	be,
	wdata,
	q
);
	integer i = 0;

	parameter WIDTH = 512;
 	parameter ADDR = 16;

	input we;
	input clk;
	input [ADDR - 1 : 0] waddr;
	input [ADDR - 1 : 0] raddr;
	input [WIDTH/8 - 1 : 0] be;
	input [WIDTH - 1 : 0] wdata;
	output reg [WIDTH - 1 : 0] q;

 
	(* ramstyle = "no_rw_check" *) logic [(WIDTH/8 - 1):0][7:0] ram[0 : (1 << ADDR) - 1] /* verilator public */; 
	always @ (posedge clk) begin
		if(we) begin
			for (i=0; i < (WIDTH/8); i = i+1)
			begin
				if(be[i]) begin
				   ram[waddr][i] <= wdata[i * 8 +: 8];
			    end
			end
		end
		q <= ram[raddr];
	end
endmodule
```

wsnyder, 2020-04-06:
Thanks for checking/asking.

A workaround is to make a bit based write enable mask then use that.  (or have your module input a bitmask.)  This will simulate crazy-faster, which is why Verilator complains.  Roughly:

        logic [WIDTH:0] bitwe;
	always_comb begin
             for (i=0; i < (WIDTH/8); i = i+1)
                  bitwe[i * 8 +: 8] = be[i];
	end

	always @ (posedge clk) begin
		if(we) ram[waddr] <= ram[waddr] & ~we | wdata & we;
	end

	always @ (posedge clk) begin
		q <= ram[raddr];
	end

I don't know if this will make an implementation difference on the FPGA; it's a common pattern so would hope not.

closed 2020-04-06 by wsnyder

oharboe, 2020-04-07:
I'm confused.... It compiles today.... I didn't intentionally try it again, I just noticed that it now compiles....

```
/**
 * From the Quartus Prime Standard Edition Handbook Volume 1
 * 
 * Based on "Example 75. SystemVerilog Simple Dual-Port Synchronous RAM with Byte Enable"
 */

module DualPortMaskedWrite(
	we, clk,
	waddr, raddr,
	be,
	wdata,
	q
);
	integer i = 0;

	parameter WIDTH = 512;
 	parameter ADDR = 16;

	input we;
	input clk;
	input [ADDR - 1 : 0] waddr;
	input [ADDR - 1 : 0] raddr;
	input [WIDTH/8 - 1 : 0] be;
	input [WIDTH - 1 : 0] wdata;
	output reg [WIDTH - 1 : 0] q;

 
	(* ramstyle = "no_rw_check" *) logic [(WIDTH/8 - 1):0][7:0] ram[0 : (1 << ADDR) - 1] /* verilator public */; 
	always @ (posedge clk) begin
		if(we) begin
			for (i=0; i < (WIDTH/8); i = i+1)
			begin
				if(be[i]) begin
				   ram[waddr][i] <= wdata[i * 8 +: 8];
			    end
			end
		end
		q <= ram[raddr];
	end
endmodule
```

wsnyder, 2020-04-07:
Probably the depth is different.  If small enough Verilator will unroll the loop for you, but it'll be slower than what I showed.

oharboe, 2020-04-07:
Gotit. Thanks!