verilator/verilator #2239: Add --build option to start C++ compilation right after verilation

yTakatsukasa, 2020-04-06:
This is byproduct of #2206 as suggested in https://github.com/verilator/verilator/pull/2206#pullrequestreview-387813514 .

Before writing code, I'd like to have the same understanding of this feature.

#### What it does
- verilator calls make at the end of veriation ( just a short cut for user)

#### Options
- `--build` : takes no arguments
- `-j` : parallelism for make.  similar to what GNU make interprets
    - `-j` : unlimited parallelism
    - `-jN` : Limit maximum process to `N`  (where N >= 1)
    - `-j N` : Limit maximum process to `N` (where N >= 1)
- `-MAKEFLAGS` _string_ :  _string_ is directly passed to make or cmake

- `--make` {cmake | gmake | make} : which command is used to build. Both gmake and make means GNU make. If `gmake` is specified, `system("gmake -f ... -C ...")` is called. If `make` is specified, `system("make -f ... -C ...")` is executed instead. If `cmake` is specified, `sustem("cmake --build ...")` is used.
#### Questions
1. In https://github.com/verilator/verilator/pull/2206#pullrequestreview-387813514,  @wsnyder wrote `number_might_be_zero`. I don't think `-j 0` is valid for GNU make. What is the desired behavior for `-j 0` ?

wsnyder, 2020-04-06:
#### Options
- `--build` : takes no arguments
- `-j` : parallelism for make.  similar to what GNU make interprets
    - `-j` : unlimited parallelism
    - `-jN` : Not allowed, will give bad option error - verilator doesn't generally support this format
    - `-j N` : Limit maximum process to `N` (where N >= 1)
    When using --build, specifies the number of jobs (commands) to run simultaneously for make/cmake.  If there  is  more  than one  -j  option,  the last one is effective.  If the -j option is given without an argument, make/cmake will not limit the number of jobs that can run simultaneously.

- `-MAKEFLAGS` _string_ :  When using --build, add the specified flag to the make/cmake command l;ine. For multiple flags either pass them as a single argument with space separators quoted in the shell (e.g. C<-MAKEFLAGS "-a -b">), or use multiple -MAKEFLAGS arguments (e.g. C<-MAKEFLAGS -l -MAKEFLAGS -k>).

- `--make` {cmake | gmake | make} : which command is used to build. Both gmake and make means GNU make. If `gmake` is specified, `system("gmake -f ... -C ...")` is called. If `make` is specified, `system("make -f ... -C ...")` is executed instead. If `cmake` is specified, `sustem("cmake --build ...")` is used.

- Should we support --jobs as a -j alias?  Propose not.
- Should we support --silent (-s) as a pass-through instead of -MAKEFLAGS -s?  Propose consider later.
- Should we support --keep-going (-k) as a pass-through instead of -MAKEFLAGS -k?  Propose consider later.

I thought "-j 0" meant same as "-j", but make doesn't do that, so it should be illegal, needs to be >= 1.

wsnyder, 2020-04-06:
One more
  -no-verilate:  When using --build, disable generation of C++/SC code, and only run make/cmake.

Need this for debug, e.g. tweaking makefiles or C code and rebuilding without having to figure out the make command Verilator did under the covers.

yTakatsukasa, 2020-04-15:
PR #2249 is merged in 18412f93220bd979e1ba71fd8c889e5229293023.
So let's close this.

closed 2020-04-15 by yTakatsukasa