verilator/verilator #2240: uniqe case report assert error when running

hdzhangdoc, 2020-04-07:
when running verilator report Error below：

[image]

there waw no assertion in code, I don't konw where the problem existed.

code is below：

 54   enum logic [1:0] { IDLE, IRQ_PENDING, IRQ_DONE} exc_ctrl_cs;
 55
 56   logic irq_enable_ext;
 57   logic [4:0] irq_id_q;
 58   logic irq_sec_q;
 59
 60 if(PULP_SECURE)
 61   assign irq_enable_ext =  ((u_IE_i | irq_sec_i) & current_priv_lvl_i == PRIV_LVL_U) | (m_IE_i & current_priv_lvl_i == PRIV_LVL_M);
 62 else
 63   assign irq_enable_ext =  m_IE_i;
 64
 65   assign irq_req_ctrl_o = exc_ctrl_cs == IRQ_PENDING;
 66   assign irq_sec_ctrl_o = irq_sec_q;
 67   assign irq_id_ctrl_o  = irq_id_q;
 68
 69   always_ff @(posedge clk, negedge rst_n)
 70   begin
 71     if (rst_n == 1'b0) begin
 72
 73       irq_id_q    <= '0;
 74       irq_sec_q   <= 1'b0;
 75       exc_ctrl_cs <= IDLE;
 76
 77     end else begin
 78
 79       unique case (exc_ctrl_cs)
 80
 81         IDLE:
 82         begin
 83           if(irq_enable_ext & irq_i) begin
 84             exc_ctrl_cs <= IRQ_PENDING;
 85             irq_id_q    <= irq_id_i;
 86             irq_sec_q   <= irq_sec_i;
 87           end
 88         end
 89
 90         IRQ_PENDING:
 91         begin
 92           unique case(1'b1)
 93             ctrl_ack_i:
 94               exc_ctrl_cs <= IRQ_DONE;
 95             ctrl_kill_i:
 96               exc_ctrl_cs <= IDLE;
 97             default:
 98               exc_ctrl_cs <= IRQ_PENDING;
 99           endcase
100         end
101
102         IRQ_DONE:
103         begin
104           irq_sec_q   <= 1'b0;
105           exc_ctrl_cs <= IDLE;
106         end
107
108       endcase
109
110     end
111   end

wsnyder, 2020-04-07:
>enum logic [1:0] { IDLE, IRQ_PENDING, IRQ_DONE} exc_ctrl_cs;

This error means the unique case isn't complete, in this case something is putting it into the 0x3 state, perhaps before reset?

wsnyder, 2020-04-16:
I presume you got this sorted out!

closed 2020-04-16 by wsnyder