verilator/verilator #2241: always_ff statement does not seem to take effect

hdzhangdoc, 2020-04-07:
A signal called destination in always_ff did not act as expected，wave dumped below：
[image]

code is below：

// Copyright 2018 ETH Zurich and University of Bologna.
// Copyright and related rights are licensed under the Solderpad Hardware
// License, Version 0.51 (the "License"); you may not use this file except in
// compliance with the License.  You may obtain a copy of the License at
// http://solderpad.org/licenses/SHL-0.51. Unless required by applicable law
// or agreed to in writing, software, hardware and materials distributed under
// this License is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
// CONDITIONS OF ANY KIND, either express or implied. See the License for the
// specific language governing permissions and limitations under the License.

/*
 * core_region.sv
 * Davide Rossi <davide.rossi@unibo.it>
 * Antonio Pullini <pullinia@iis.ee.ethz.ch>
 * Igor Loi <igor.loi@unibo.it>
 * Francesco Conti <fconti@iis.ee.ethz.ch>
 */

import apu_package::*;

// USER DEFINED MACROS to improve self-testing capabilities
`ifndef PULP_FPGA_SIM
  `define DEBUG_FETCH_INTERFACE
`endif
//`define DATA_MISS
//`define DUMP_INSTR_FETCH

module core_region
#(
  // CORE PARAMETERS
  parameter int    CORE_ID                = 0,
  parameter int    ADDR_WIDTH             = 32,
  parameter int    DATA_WIDTH             = 32,
  parameter int    INSTR_RDATA_WIDTH      = 32,
  parameter bit    CLUSTER_ALIAS          = 1'b1,
  parameter int    CLUSTER_ALIAS_BASE     = 12'h000,
  parameter int    REMAP_ADDRESS          = 0,
  parameter int    DEBUG_HALT_ADDR        = 32'h0000_0000,
  parameter int    APU_NARGS_CPU          = 2,
  parameter int    APU_WOP_CPU            = 1,
  parameter int    WAPUTYPE               = 3,
  parameter int    APU_NDSFLAGS_CPU       = 3,
  parameter int    APU_NUSFLAGS_CPU       = 5,
  parameter bit    ADDREXT                = 1'b0,
  parameter        FPU                    = 0,
  parameter        FP_DIVSQRT             = 0,
  parameter        SHARED_FPU             = 0,
  parameter        SHARED_FP_DIVSQRT      = 0,
  parameter bit    DEM_PER_BEFORE_TCDM_TS = 1'b0
`ifndef SYNTHESIS
  ,
  parameter string L2_SLM_FILE            = "./slm_files/l2_stim.slm",
  parameter string ROM_SLM_FILE           = "../sw/apps/boot/slm_files/l2_stim.slm"
`endif
  )
(
  input  logic            clk_i,
  input  logic            rst_ni,
  input  logic            init_ni,

  input  logic [3:0]      base_addr_i, // FOR CLUSTER VIRTUALIZATION
  input  logic [5:0]      cluster_id_i,

  input  logic            irq_req_i,
  output logic            irq_ack_o,
[... 367 lines cut ...]
    end
  end
`endif

`ifdef DATA_MISS
  logic data_hit;
  logic req;
`endif
  logic reg_cache_refill;

  always_ff @(posedge clk_i , negedge rst_ni)
  begin
    if ( rst_ni == 1'b0 ) begin
      reg_cache_refill <= 1'b0;
    end
    else begin
      if (instr_req_o)
        reg_cache_refill <= 1'b1;
      else if(instr_r_valid_i && !instr_req_o)
        reg_cache_refill <= 1'b0;
    end
  end
//synopsys translate_on

endmodule

wsnyder, 2020-04-07:
I'm sorry you have problems but we don't have time to do detailed debugging for you. Please assume we always need a small self-checking test case to debug problems, and delay filing an issue until you have one.

wsnyder, 2020-05-07:
Please reopen or refile if you can provide a small test case.

closed 2020-05-07 by wsnyder