verilator/verilator #2242: Unsupported: Verilog 1995 gate primitive: pmos

hdzhangdoc, 2020-04-08:
When compiling，I met an Error：

[image]
How can I eliminate this error?

wsnyder, 2020-04-08:
Gate primitives won't be supported, you need to code it a different way not using gate primitives.

Also as I've asked several times, please post *text* not images when the subject is text.

closed 2020-04-08 by wsnyder

hdzhangdoc, 2020-04-08:
OK，thank you for your help. And I would like to ask if the --un-box option can be used to compile, is the simulation result correct?

wsnyder, 2020-04-08:
I assume you mean --bbox-*. You cannot use these and get a correct simulation.  Given you have these soft of constructs and seem to be having other issues please try to use Icarus Verilog instead.