verilator/verilator #2243: Code in always_ff behaves incorrectly

hdzhangdoc, 2020-04-08:
We found a possible bug in the process of locating a problem. The state transition of a state machine is in the always_ff statement，Assigned from NS to CS, but from the waveform it looks like NS is assigned to CS. This is the opposite of running on vcs.

Part of the code is as follows：

206    always_ff @(posedge clk, negedge rst_n)
207    begin
208       if(~rst_n)
209       begin
210           CS                       <= DISABLED_ICACHE;
211           fetch_addr_Q             <= '0;
212           fetch_req_Q              <= '0;
213           fetch_way_Q              <= '0;
214
215           counter_FLUSH_CS         <= '0;
216
217       end
218       else
219       begin
220           **CS <= NS;**
221           counter_FLUSH_CS         <= counter_FLUSH_NS;
222
223           if(save_fetch_way)
224             fetch_way_Q              <= fetch_way_int;
225
226           if(enable_pipe)
227           begin
228                fetch_req_Q    <= 1'b1;
229                fetch_addr_Q   <= fetch_addr_i;
230           end
231           else  if(clear_pipe)
232                 begin
233                      fetch_req_Q <= '0;
234                 end
235
236       end
237    end

wave generated by Verilator ：
[image]

wave generated by VCS：
[image]

hdzhangdoc, 2020-04-08:
Sorry, there is a place where the code may be related to this, I add it below 

272 always_comb
273 begin
274    TAG_req_o          = '0;
275    TAG_we_o           = 1'b0;
276    TAG_addr_o         = fetch_addr_i[SET_ID_MSB:SET_ID_LSB];
277    TAG_wdata_o        = {1'b1,fetch_addr_Q[TAG_MSB:TAG_LSB]};
278
279    DATA_req_o         = '0;
280    DATA_addr_o        = fetch_addr_i[SET_ID_MSB:SET_ID_LSB];
281    DATA_wdata_o       = refill_r_data_i;
282    DATA_we_o          = 1'b0;
283
284    fetch_gnt_o        = 1'b0;
285    fetch_rvalid_o     = 1'b0;
286    fetch_rdata_o      = refill_r_data_i; //FIXME ok for AXI 64 and 32bit INSTR
287
288    refill_req_o       = 1'b0;
289    refill_addr_o      = fetch_addr_i;
290    fetch_way_int      = '0;
291
292
293
294    save_fetch_way          = '0;
295
296    enable_pipe             = 1'b0;
297    clear_pipe              = 1'b0;
298
`**299    NS                      = CS;**`
300    update_lfsr             = 1'b0;
301
302    cache_is_bypassed_o     = 1'b0;
303    cache_is_flushed_o      = 1'b0;
304
305    counter_FLUSH_NS        = counter_FLUSH_CS;
306
307    flush_set_ID_ack_o      = 1'b0;
308
309    hit_counter_enable      = 1'b0;
310    miss_counter_enable     = 1'b0;
311
312    case(CS)
313
314       DISABLED_ICACHE:
315       begin
316          flush_set_ID_ack_o  = 1'b1;

wsnyder, 2020-04-08:
I'm sorry you have problems but we don't have time to do detailed debugging for you. Please assume we always need a small self-checking test case to debug problems, and delay filing an issue until you have one.

hdzhangdoc, 2020-04-09:
I'm sorry it's very difficult to provide a self-checking test case to debug problems，I tried to compile only this file but is acted rightly. This problem only appears when compiling this file with other files on my system.

wsnyder, 2020-04-09:
Ok, if you get it to a small case, feel free to repoen.

closed 2020-04-09 by wsnyder