verilator/verilator #2247: Should --output-split and parallallel make be the default?

gezalore, 2020-04-10:
Currently the default make rule for verilated models compiles the C++ via the *__ALL*.cpp files which include all code into a single invokation of the compiler (well, one invokation for fast path code with optimization and one invocation for slow path code with less optimizaiton). This makes the compilation of the fast path code the long pole when building the verilated mode. There is great build-time improvement to be gain from using parallel build (`make -j`). For that to work reasonably you need the `--output-split` Verilator option, which in recent versions also sets the requied `VM_PARALLEL_BUILDS=1` when invoking Make to build the mode (#2185).

Should we turn  `--outptu-split` on by default, so `make -j` does what is expected out of the box? 

Note this question also came up in #2245 

If we do turn `--output-split` on by default, the question is with wath value? I personally find it difficult to figure out the value I should provide to it, so usually end up going with some semi wild guess after a couple iterations to check basic build time and run-time speed when I'm setting up a project, but rarely have the bandwidth to revisit it afterwards.

wsnyder, 2020-04-10:
I had a theory that making --output-split the default might slow down small tests and it proved out
with an experiment. If I run "make test" with it on, the time goes up from 92 CPU minutes to 130 CPU minutes total.  The problem is the cost of doing all of the headers over-and-over exceeds the advantage.  I think the optimal default is probably to not split/parallel until a certain size is reached, then split.

How to determine the number for output-split is an interesting question.

I don't think we should just split into "N" pieces as then when something small in the model changes it might split differently - we need to go in the other direction of more stability to make ccache better.

Ideally we could figure out an algorithm that would work everywhere, worth some experiments.

It's perhaps better to have nearly any default versus the current massive file for large designs.

Maybe precompiled headers would help both cases, maybe to the point of making splitting relatively finely generally good?

gezalore, 2020-04-11:
So, if the only problem is the headers, how about getting rid of them completely? I think most of the dependency on the headers is due to the symbol table, but the nice thing about that is that we know  it's structure exactly, so we could just replace it with an array of bytes and use static offsets (which are now allocated by Verilator instead of the C compiler). I'm cringing a bit at the though of debugging what Verilator would out at that point, but then we could emit getters and setters for the symbols (which just access the array at the static offset) and use them in a debug mode build. Maybe it could be implemented so the last thing during compilation would be to inline the getters and setters and if we did so, dont include their definition? I think this would make practically all important functions individually compilable with no overhead so could emit one function per file even. It would also certainly help with #2140. I wonder if there is something in the C standard that would make the above inefficient as it must use a lot of `reinterpret_cast`..

gezalore, 2020-04-11:
Alternative to the mega byte array would be to use a union like this, assuming the aliasing rules don't make it useless:
```
union Vtop_symbols {
  CData   C[size];
  SData   S[size/2];
  IData   I[size/4];
  QData   Q[size/8];
  double  dbl[size/8];
  ....
}
```

wsnyder, 2020-04-11:
The header compilation cost I was referring to is verilated*.h. The self tests tend to have very small symbol tables so the sym.h's don't cost much time in them.

In a "real" design the Syms I suspect are a contributor (I haven't measured) to compile time, but perhaps more critically cause recompiling everything on any signal add/remove.  #2140 discusses a possible proposal to reduce header dependencies and assert everything is layed out reasonably without strange accessors.  Suggest we move that part of the discussion there.

gezalore, 2020-04-15:
One thing I was wondering about, for other reasons as well, but would also help small tests, is whether we should build a `libverilated.a` at the time Verilator is built, that contains all the things that are now in `VM_GLOBAL*` (i.e.: *.c/*.cpp files in `include/`) and is currently built with the model. If you are using `VM_PARALLEL_BUILDS` (which big projects do) these are compiled on their own anyway so using a pre-compiled library would not make a difference. This would help the small test cases as they would not have to re-build `verilated.cpp` in particular. In addition, and I would argue more importantly, we could just build them with max optimization once, which would speed up things like tracing (the formatting code is now in the cpp files) a lot if the model itself is compiled with -O0 for faster build time. Some of the cpp files in `incldue/` have dependencies on compile time defines, but these seem to be very mild, like these which would just be overriden automatically due to link order if the user provides one:
https://github.com/verilator/verilator/blob/f3308d236b5ae329241f4b16c3b39c282c351d9a/include/verilated.cpp#L92-L98
Having skimmed the rest of the dependencies on model-compile-time values, they don't seem hard to fix. I would be happy to prototype this in a PR unless there is some reason I cannot see why these files must be built with the model.

wsnyder, 2020-04-15:
Assuming the build rules are right and ccache is setup right, then I don't see an advantage to a library. For example the "test regress" currently shows ccache hits on other tests's verilated usages.

The downside I see is as you note, namely picking the right library to deal with different debug settings, multithreading or not multithreading, etc.  Also the target OS/compiler may be different from the one used to build Verilator, so we'd have to figure that out at verilator install time.

If we used a Verilated shared objects I think there might be system benefits, but don't think we're likely to do that for performance reasons.

So I guess I generally see using .a as significant work with little benefit with CCache now on by default, if we can also turn on parallel by default.  In contrast making ccache works efficiently (e.g. #2140) is of wide benefit.

gezalore, 2020-04-15:
So should we just build all VM_GLOBAL speratately (i.e.: not via the __ALL.cpp), with -O3 all the time? That does not seem harmful under your assumptions.

> Also the target OS/compiler may be different from the one used to build Verilator, so we'd have to figure that out at verilator install time.

I don't think that is quite right. The build of the model already depends on the result of configure via verilated.mk, and is full of target specific flags, so if you want to build on a different OS/compiler, you will need a corresponding verilator installation for that target and use that as VERILATOR_ROOT when building the model, even if you ran Verilator elsewhere. Having a library actually would just simplify that as all you would have to have installed on the remote end is the library, but not Verilator itself. In fact we could also build the precompiled headers at the same as building the lib. 

BTW, other than the speed of the test suite, is there anything else preventing us from always using VM_PARALLEL_BUILDS=1? I think anyone else who just wants to build a model, and not ~3000 models at the same time like we do for the test suite would be better of with VM_PARALLEL_BUILDS=1 being the default.

wsnyder, 2020-04-15:
>So should we just build all VM_GLOBAL speratately (i.e.: not via the __ALL.cpp), with -O3 all the time? 

Yes, it would probably be reasonable to use -Os for this.

>BTW, other than the speed of the test suite, is there anything else preventing us from always using VM_PARALLEL_BUILDS=1?

I think the test suite is potentially representative of people using Verilator on small often-churning designs, so would be reluctant to have the tests e.g. override default behavior.  I think simply setting the default based on some minimum size metric is reasonable -- probably should be a new flag but likely the default setting is the same as that for output-split, that is if output split never causes a split, then likely there's no point in parallel.

However as you point out, always using VM_GLOBAL as a separate compile, regardless of PARALLEL_BUILDS is probably the right thing.

The way to benchmark this is to flush the ccache "ccache -C" and "time make test".  I'm ok with small losses here if need be.

gezalore, 2020-04-15:
Ok, makes sense. VM_GLOBAL is actually already separate, even with the parallel build.

You mentioned -Os should be the default, but these are actually commented out, so the default is no optimization if the user just invokes the generated makefile.  I assume this is not intentional so I will flip these on too.
https://github.com/verilator/verilator/blob/9b8aebb00cd7243a8806994257e1ba290df46927/include/verilated.mk.in#L90-L94

wsnyder, 2020-04-15:
Yes, we probably should comment in OPT_FAST. This is a case though where we will want to do something for the test_regress driver to override it, or it will get much slower.

gezalore, 2020-04-18:
I ran some experiments using precompiled headers. I bundled all headers that the particular model used from `include/` (and only the ones actually reqired) into a verilated_all.h, precompiled that in 2 falvours for OPT_FAST and OPT_SLOW, then built the model against those, using VM_PARALLEL_BUILDS=1.

On Clang, it was not bad. The total CPU time for the test suite went up 3% compared to building with the current serial mode (VM_PARALLEL_BUILDS=0). For reference, on my system, building the test suite with VM_PARALLEL_BUILDS=1 is 17% more CPU time, so the precompiled headers brought you back to almost the same level in parallel mode as you were in serial mode.

On GCC, it was quite bad. Parallel mode without precompiled headers is +16% compared to serial mode. Parallel mode with precompiled headers is +25%... GCC takes longer to build the precompiled header, which is not offset by it's reuse. The precompield header itself is also huge, ~25MB per test, so I filled up the 80GB spare space on my disk when I got through about half the test suite on the first try..

What makes things worse, is that ccache will only use cached precompiled headers if it can do so in direct mode (i.e.: all preprocessor flags must match), which makes it useless for the test suite as all tests have different -D options. (Acutally, newer versions will cache it for GCC, but not Clang, in preprocessed mode, but still only if all options are the same..)

gezalore, 2020-05-26:
This is mostly done thanks to #2365, the only thing left in this issue we discussed is whether we should set OPT_FAST to -Os by default which I think we agreed we should (but override it in test_regress for speed). If that is still the case I will sneak that in shortly.

wsnyder, 2020-05-26:
Sounds good, but also separate commit please.

wsnyder, 2020-05-26:
And put in Changes - but you probably already were going to!

gezalore, 2020-05-27:
Finished in #2374

closed 2020-05-27 by gezalore