verilator/verilator #2248: Code specialization for quasi static signals

gezalore, 2020-04-10:
This is to discuss a strategy for potential model performance improvement. The idea is to use code specialization for quasi static signals. The obvious example is reset, which in the SweRV EH1 testbench is generated with:

```
assign rst_l = cycleCnt > 5;
assign porst_l = cycleCnt >2;
```

This means pretty much after the first few cycles, all resets will be effectively constant, but Verilator emits code like this all over the place:

```
vlTOPp->tb_top__DOT__rvtop__DOT__swerv__DOT__ifu__DOT__mem_ctl__DOT__tagv_mb_ff
        = ((IData)(vlTOPp->tb_top__DOT__rvtop__DOT__core_rst_l)
             ? (IData)(vlTOPp->tb_top__DOT__rvtop__DOT__swerv__DOT__ifu__DOT__mem_ctl__DOT__tagv_mb_in)
             : 0U)
```

We could just take the whole code, and emit 2 versions of it, with the reset signal inlined with the 2 possible constant values, and then check at the top of `eval` based on the actual value of the signal which version to call. This increases satic code size but the size of the frequently used hot code is actually reduced. Even if they are easy to predict branches, not having masses of them flowing through your processor frontend can be beneficial.

Reset is just one example where this can help. it is generally applicable to any signal that is expected to stay constatn for long periods of time but influences execution on a fine grained level in a lot of places.

There are quite a couple of challenges in the above, including:
- How do we know what signals are quasi static? we can probably guess reset, using a heuristic but user annotation might be necessary for others.
- Quasi static signals are often combined before their use (e.g.: the 2 resets in the testbench of EH1 are combined before used by the design), so we need to propagate the property through logic, i.e.: a signal is quasi static if all its inputs are quasi static.
- Must avoid the previous point triggering a combinatorial explosion, so some heuristics are needed on how many specializations are beneficial, but how many are too many.
- Can harm build time as you now have loads more static code to compile. This could be aleviated by recognising some of the paths (e.g.: reset asserted) as 'slow' code so they can be compiled without optimization.

wsnyder, 2020-04-10:
I'd be interested in some experiments on SweRV for this.  I have experimented in the past on other models by manually copying the code and search-and-replacing "reset" with e.g. 0 and hacking up the values in registers to be correct post-reset.  I was surprised I didn't get enough performance to make it seem worthwhile (< 10% on something other than SweRV); basically large statements and such use most the time and the resets just don't matter much as the signals stick in the cache and the branch predictor worked well.  For SweRV we'd want to compare to the 8% gain for profile driven compiler feedback.

I suspect we'd want to handle the static detection by requiring a toggle coverage run, then feeding the data back into Verilator, similar to how the compiler does profile-driven optimization. Verilator would compare the toggle rate to the number of usages of the signal - only those with very low usage and high fanout and maybe some other heuristics would qualify.  Plus of course manual overrides.

wsnyder, 2023-09-26:
Closing, as unlikely to get investigated or implemented for a long time.

closed 2023-09-26 by wsnyder