verilator/verilator #2255: Failed to compile with multithreading with linker error

mmatzev, 2020-04-12:
I try to compile a design similar to the pulpissimo SoC from github.com/pulp-platform/pulpissimo. It works well with verilator as long as single-threading in selected.

If I add the option "--threads 4" I get the following error message:
/usr/bin/ld: _ZN9Verilated3t_sE: TLS reference in Vdimad01_digital_top__ALL.a(Vdimad01_digital_top__ALLcls.o) mismatches non-TLS definition in verilated.o section .bss
/usr/bin/ld: Vdimad01_digital_top__ALL.a: error adding symbols: bad value
collect2: error: ld returned 1 exit status

I used the Windows-Subsystem for Linux with the Debian distribution.

wsnyder, 2020-04-12:
A google search of the error (minus verilator specifics) suggests:

" If you are compiling an application under Linux and get an error message like:
/usr/bin/ld: errno: TLS definition in /lib/libc.so.6 section .tbss mismatches non-TLS reference in ...
The fix is quite simple: explicitly add #include <errno.h> to the offending file and then recompile.

So this would be to verilated.h.  If this works please consider submitting a pull request to fix it.

If not, maybe there are other online suggestions.

mmatzev, 2020-04-12:
I added the include in verilated.h as suggested and than the options "--threads" works.
But than the compile does not work any more without the option "--threads".
I tried also to place it in the "ifdef VL_THREADED" but it didn't help. 

My change in my own repository:
Index: /home/vkr/ki_predict/iss_ki_predict/trunk/tools/wsl-debian/verilator/4.032/share/verilator/include/verilated.h
===================================================================
--- /home/vkr/ki_predict/iss_ki_predict/trunk/tools/wsl-debian/verilator/4.032/share/verilator/include/verilated.h     (revision 17)
+++ /home/vkr/ki_predict/iss_ki_predict/trunk/tools/wsl-debian/verilator/4.032/share/verilator/include/verilated.h     (working copy)
@@ -31,6 +31,7 @@
 #include <cstdio>
 #include <cstdlib>
 #include <cstring>
+#include <errno.h>
 // <iostream> avoided to reduce compile time
 // <map> avoided and instead in verilated_heavy.h to reduce compile time
 // <string> avoided and instead in verilated_heavy.h to reduce compile time

drichmond, 2020-04-15:
I also ran into this problem, but the solution above doesn't fix it

```/opt/rh/devtoolset-8/root/usr/libexec/gcc/x86_64-redhat-linux/8/ld: _ZN9Verilated3t_sE: TLS reference in Vtop__ALL.a(Vtop__ALLcls.o) mismatches non-TLS definition in ../verilated.o section .bss```

Still looking into other solutions

drichmond, 2020-04-15:
Ignore my previous comment for now, not seeing it in some situations and I am seeing it in others. Will work to find out more.

wsnyder, 2020-05-27:
Any feedback on if you got Verilator multithreading working?  Would like to close this issue out. Thanks

drichmond, 2020-05-27:
Yeah, I got it working. I needed to use dev tools-8 on Centos/RHEL

wsnyder, 2020-05-27:
Great!

closed 2020-05-27 by wsnyder

riteme, 2021-02-09:
I ran into this problem when I tried to add either `--threads 1` or `--trace-threads 1` to Verilator.

I found that I just forgot to add `-DVL_THREADED` when compiling `verilated.cpp` into `verilated.o` by G++.