verilator/verilator #2256: Failed to build verilator with some DPI function on windows (msys2)

mmatzev, 2020-04-12:
I tried to build a verilator model on msys2 (windows). This model has 2 modules using DPI.
a) a JTAG interface, which calls C-functions from SystemVerilog
b) a Memload function, which allows to load the memory from C

The compilation works on CentOS and Debian. On Msys2 it works with the JTAG interface but it 
does not compile with the memload function. 

I get the error message: 
g++    digital_top.o jtagdpi.o spi_slave.o tb.o verilated.o verilated_dpi.o verilated_vcd_c.o Vdimad01_digital_top__ALL.a    -o Vdimad01_digital_top -lm -lstdc++
/usr/lib/gcc/x86_64-pc-msys/9.3.0/../../../../x86_64-pc-msys/bin/ld: digital_top.o:digital_top.cpp:(.text+0x182): undefined reference to `WriteVmemToMem'
digital_top.o:digital_top.cpp:(.text+0x182): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `WriteVmemToMem'
collect2: Fehler: ld gab 1 als Ende-Status zurück

WriteVmemToMem is the C-function which triggers the SystemVerilog function. See attached code.

Any idea, why this is a problem (only) on windows?

P.S.: I used msys2 and not mingw, as this is not supported by the JTAG DPI function, which uses a linux style network implementation
[memload.zip](https://github.com/verilator/verilator/files/4466334/memload.zip)
[sp_ram.zip](https://github.com/verilator/verilator/files/4466337/sp_ram.zip)

wsnyder, 2020-04-12:
I suspect the object files were compiled with different flags.  Try putting all of the .c/.cpp files on the verilator command line and it will add them to the verilated make with consistent flags.

mmatzev, 2020-04-12:
All c and cpp files are included in the verilator command line. 
I tried also to rewrite the memload.cpp function in plain c, but it didn't help.
 
I attached the generated make file. Maybe an compile option is wrong!?
[Vdimad01_digital_top.zip](https://github.com/verilator/verilator/files/4467028/Vdimad01_digital_top.zip)

wsnyder, 2020-04-12:
>        -std=gnu++0x 

This must be a 9+ year old compiler, is there something more recent you can use?  Might be the problem with the other issue too.

mmatzev, 2020-04-13:
I just kept the command line as it was. 
Now, I removed this option. The new Makefile is attached. Unfortunately I got the same error message. (By the way: the gcc is Version 9.3.0

g++    digital_top.o jtagdpi.o memload.o spi_slave.o tb.o verilated.o verilated_dpi.o verilated_vcd_c.o Vdimad01_digital_top__ALL.a    -o Vdimad01_digital_top -lm -lstdc++
/usr/lib/gcc/x86_64-pc-msys/9.3.0/../../../../x86_64-pc-msys/bin/ld: memload.o:memload.c:(.text+0x17): undefined reference to `__imp_svGetScopeFromName'
memload.o:memload.c:(.text+0x17): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp_svGetScopeFromName'
/usr/lib/gcc/x86_64-pc-msys/9.3.0/../../../../x86_64-pc-msys/bin/ld: memload.o:memload.c:(.text+0x45): undefined reference to `__imp_svSetScope'
memload.o:memload.c:(.text+0x45): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp_svSetScope'
/usr/lib/gcc/x86_64-pc-msys/9.3.0/../../../../x86_64-pc-msys/bin/ld: memload.o:memload.c:(.text+0x65): undefined reference to `__imp_svSetScope'
memload.o:memload.c:(.text+0x65): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp_svSetScope'
collect2: Fehler: ld gab 1 als Ende-Status zurück

[Vdimad01_digital_top.zip](https://github.com/verilator/verilator/files/4469569/Vdimad01_digital_top.zip)

mmatzev, 2020-04-13:
A small remark 
I'm not in compiling C code but maybe this warning gives a hint?

g++  -I.  -MMD -I/home/vkr/ki_predict/trunk/tools/msys2-windows/verilator/4.032/share/verilator/include -I/home/vkr/ki_predict/trunk/tools/msys2-windows/verilator/4.032/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -DVL_THREADED -std=gnu++14  -c -o digital_top.o /home/vkr/ki_predict/trunk/rtl/verilator/digital_top.cpp
In Datei, eingebunden von ./Vdimad01_digital_top__Dpi.h:8,
                 von ./Vdimad01_digital_top.h:13,
                 von /home/vkr/ki_predict/trunk/rtl/verilator/digital_top.h:16,
                 von /home/vkr/ki_predict/trunk/rtl/verilator/digital_top.cpp:13:
/home/vkr/ki_predict/trunk/tools/msys2-windows/verilator/4.032/share/verilator/include/vltstd/svdpi.h:123:22: Warnung: Attribut »dllimport« wird ignoriert [-Wattributes]
  123 | XXTERN typedef void* svScope;
      |                      ^~~~~~~
/home/vkr/ki_predict/trunk/tools/msys2-windows/verilator/4.032/share/verilator/include/vltstd/svdpi.h:126:22: Warnung: Attribut »dllimport« wird ignoriert [-Wattributes]
  126 | XXTERN typedef void* svOpenArrayHandle;

wsnyder, 2020-04-13:
It's a hint that you are a German speaker ;)

Actually that might be it, try defining "-DDPI_DLLISPEC -DDPI_DLLESPEC". These defines come from IEEE and are often not very compiler appropriate.

mmatzev, 2020-04-14:
Dear, the compile warnings disappeared with the additional flags.
The problem remains.

A closer look with "objdump.exe -d verilated_dpi.o  | less" showns that there is svSetScope defined but not "__imp_svSetScope" as written in the error message.

mmatzev, 2020-04-14:
Sorry, my fault. After cleaning the build dir and running verilator again it works with your options.
A little bit interesting - It works only if the build dir is cleaned before running verilator.

closed 2020-04-14 by mmatzev