verilator/verilator #2258: SystemC phase callback based tracing.

gezalore, 2020-04-13:
After further analysis of #1307 and #1445, to me it actualy seems like that "phase callback based tracing" is a new experimental feature in SystemC. #1445 describes an issue about trace not apperaing when using SystemC at all. This is due to the user building that version of SysemC with
`-DENABLE_PHASE_CALLBACKS_TRACING=ON` (which is the default in its CMake config), and results in the calls to the `sc_trace_file::cycle` mehtods being compiled out of the SC kernel. My interpretation is that this is not a bug, and the proper solution is to use a `before_timestep` phase callback for tracing in this configuration, which happens exactly at the point when the `sc_trace_file::cycle` was called with the `delta_cycle` argument `false`. This is my conjecture based on reading the code of SystemC 2.3.3 (see sc_sim_context.cpp). Doesn't feel like a priority issue to me, but I thought it's worth remembering.

#2257 Adds a not to the manual explaining this.

wsnyder, 2020-04-13:
Makes sense, suggest when #2257 pushes we can close this as opposed to leaving it as a feature bug.

gezalore, 2020-04-13:
So, I'm happy to close it as won't fix, but just o be sure, my point was that it might be possible to update `verilated_vcd_sc.{h,cpp}` so it works wiht the new SystemC thing. It's also OK not to, I personally never use SC.

wsnyder, 2020-04-13:
Understood, thanks. We want to keep SystemC working, but isn't really getting much other attention as is a much smaller user base.

Note to future readers - if you hit this, please file and issue or ideally consider a pull request to fix it, as our thinking was this wasn't in high demand otherwise. Thanks

gezalore, 2020-04-13:
Closing as discussed as this is now described in the manual with the merge of #2257 wa

closed 2020-04-13 by gezalore