verilator/verilator #2262: Asserting internal signals does propagate - How can propagation be achived

Thonner, 2020-04-16:
I'm am working on an emulator for a processor using verilator. 
I need to force some internal signals. In order to get access to internal signals from my verilator harness, I have given ventilator the --public argument. What i see in my VCD dumb is that only the signal that I directly force change value. This value is not propagated through the circuit. In the screenshot is an example - I have forced pcReg to 3FF..., which is directly connected to module output io_fedec_pc, which does not change its value.

Is there a special operation that I need to perform in order for forced internal signal values to propagate further? 

I appreciate the help - Thank you

[image]

wsnyder, 2020-04-16:
Perhaps you didn't call eval() after depositing the value?

This is also suggested by the value going away at the next edge. When you deposit the value you will need to do something to prevent the normal clocking from overriding it.  That is, it's not a "force" statement but rather a change of storage value that the next "normal" flop data will override.

BTW an alternative is to have a register just for overriding and an enable for overriding, e.g.

```
reg [...] normal_pc;;
reg [...] deposit_pc;  /* verilator public_flat_rw @(posedge clk) */;
reg deposit_enable  /* verilator public_flat_rw @(posedge clk) */;
always_ff
    normal_pc <=  ... normal code which probably recirculates pc;
wire [...] pc = deposit_enable ? deposit_pc : normal_pc;
```   

Your testbench then puts a value in deposit_pc and sets deposit_enable.

Also note --public can cause other problems with e.g. clocks, it's better to use the metacomments (or a vlt file) to only expose those signals needed.

Thonner, 2020-04-16:
Thank you for your input!

I have a function for doing ticks, which cals eval():
`
  void tick(void)
  {
    m_tickcount++;
    c->eval();
    if (trace) {
      c_trace->dump(10*m_tickcount);
    }
    c->clock = 0;
    c->eval();
    if (trace) {
      c_trace->dump(10*m_tickcount+5);
    }

    c->clock = 1;
    c->eval();

    if (trace) {
      c_trace->dump(10*m_tickcount+10);
      c_trace->flush();
    }
`
In main of my harness im depositing the internal signal and call tick afterwards. To me this should be okay. So could there be an other reason why the deposited value do not propagate?

It is fine that the next rising edge will override the register content, I just want the deposited data to propagate further down the circuit the cycle that it is being deposited.

Implementing your "deposit-on-enable register" could be done. But I like to avoid this. This processor platform also is used for teaching a computer architecture course. Using the special register means pulling out these signals from internal modules to the top module, for emulation purposes only. This could cause confusion for the student reading the code, which I would like to avoid if possible.

Yes I have read that you shouldn't use --public several places ([Accessing Signals in Verilator Models ](url)). I would like to avoid this, as it also makes the emulation run slower. But it is not clear to me where I should put the **verilator public** comment. Is it in the signal declaration in my Verilog file? 

Thank you again. I am new to verilator but are excited to become proficient with the tool!

wsnyder, 2020-04-16:
--public doesn't really change the edge events happen on, so likely your signal only normally changes on posedges, so your force will take effect downstream the next posedge, but on that posedge the logic will also get the new flop value.

Try using /*verilator public_flat_rw @(negedge clk)*/  that is, saying your deposit needs to propagate on negedges, and do the deposit before you propage clk falling with eval().  You can also use control scripts if you don't want to edit rtl.

`public_flat_rw -module "sub" -var "pcReg" @(negedge t.clk)
`

Thonner, 2020-04-23:
Thank you for your help!

I have had success with meta comments. I needed to install verilator from the current stable release from the Github repository in order to have in the configuration file. The version I had installed through apt-get did not recognize the public_flat_rw identifier in a configuration file.

closed 2020-04-23 by Thonner