verilator/verilator #2263: dearrayAll segfault on sliced interface

engr248, 2020-04-17:
Creating new issue after recommendation from #1430 thread.

The attached test_regress format files segfault somewhere in dearrayAll on verilator v4.032.

Running with --debug --gdb and running `call nodep->fileline()->ascii()` at the fault shows that taking a slice of a 1D array of interfaces and assigning it to a submodule is the faulty line (noted in comments in the test code).

[t_dearray_segfault.zip](https://github.com/verilator/verilator/files/4494332/t_dearray_segfault.zip)

engr248, 2020-04-17:
Follow-up: as a result of distilling down my code into the test case, I had an idea for a workaround. I created a temporary array of interfaces that's the exact size needed for the module port and then have a for loop that just assigns the interfaces between the original array (that was previously being sliced) to/from the temporary array, which is then connected to the port. This seems to allow verilator to continue and I'm back to simulating and debugging the function of the design rather than the segfault.

wsnyder, 2020-04-17:
Thanks for the test case.  You diagnosed the correct problem & workaround.

Fixed in git towards 4.034.

closed 2020-04-17 by wsnyder