verilator/verilator #2267: Regression with hierarchical references used inside a function

jpallister, 2020-04-19:
Hi,

I'm getting a problem when running some code with hierarchical references, which results in an internal error:
```
$ verilator --cc failure.sv
%Error: Internal Error: failure.v:19:37: ../V3LinkDot.cpp:2304: Couldn't resolve inlined scope 'accessor_block' in: mod_a.u_a_mon.accessor_block
   19 |          bit read_x = mod_a.u_inner.x;
      |                                     ^
                        ... See the manual and https://verilator.org for more assistance.
```

```verilog
module mod_a;
   mod_inner u_inner;
   mod_a_mon u_a_mon;
endmodule : mod_a

module mod_inner;
   logic x;
endmodule : mod_inner

module mod_a_mon;
   function void accessor;
      begin : accessor_block
         bit read_x = mod_a.u_inner.x;
      end
   endfunction
endmodule
```

The above code was reduced from an internal design which is currently using verilator version 4.010 (and attempting to update to version 4.032). Git bisect suggests that the issue was introduced in #2176. 

Thanks,
James

wsnyder, 2020-04-20:
Confirmed - perhaps you would be willing to look into why this fails and attempt a patch?

Otherwise I will look into it within a few days.

jpallister, 2020-04-21:
I'm happy to have a go at fixing this.

I've had a look into what is going wrong:
 - the error reports "mod_a.u_a_mon.accessor_block", which seems to be missing the scope of the function ("accessor")
 - it looks like AstVarXRef nodes under a function are "rescoped" (in [V3Begin](https://github.com/verilator/verilator/blob/d4f7f5297a0efca79bfa8f2aed6bb6b6be0206a3/src/V3Begin.cpp#L213)) but in this case m_namedScope is "accessor_block", rather than including the function name (perhaps it should be "accessor.accessor_block" ?)

I notice that a lot of the visit functions have special cases for functions, so I'll attempt to do something similar. Does this sound like it is along the right lines?

wsnyder, 2020-04-21:
That seems right, looking at the broken fix you mention should also help isolate it further.  (Or maybe that's how you got there!)

wsnyder, 2020-09-27:
Fixed with #2572, thanks for looking into this, the fix ended up after debugging much simpler than I had feared!

closed 2020-09-27 by wsnyder