verilator/verilator #2271: UNable to compile on pwd

ajwhooley, 2020-04-20:
How may we help - what is your question?

ajwhooley, 2020-04-20:
Hi,
    I am trying to compile the Verilator source code and have tried both RH6.8 and CentOS7 7.6. Both fail for similar reasons:
/usr/include/FlexLexer.h:121: error: candidate is: virtual yy_buffer_state* V3LexerBase::yy_create_buffer(std::istream*, int)
V3Lexer_pregen.yy.cpp: In member function ‘virtual yy_buffer_state* V3LexerBase::yy_create_buffer(std::istream*, int)’:
V3Lexer_pregen.yy.cpp:8801: error: invalid conversion from ‘void*’ to ‘std::istream*’
V3Lexer_pregen.yy.cpp:8801: error:   initializing argument 1 of ‘virtual yy_buffer_state* V3LexerBase::yy_create_buffer(std::istream*, int)’
V3Lexer_pregen.yy.cpp: At global scope:
V3Lexer_pregen.yy.cpp:8835: error: prototype for ‘void V3LexerBase::yy_init_buffer(yy_buffer_state*, std::istream&)’ does 

Is there something obvious I am missing/doing wrong?

Alan

wsnyder, 2020-04-20:
Sorry you are having problems.  This is almost certainly due to a very old & broken version of flex, please install a recent version.  Alternatively, googling the error message gives this
https://stackoverflow.com/questions/45382360/flex-lexer-error-to-build-rapidnet-stimulator

Note that RH 6 is 10 years old, stopped getting upgrades in 2017, and is end of life this year.  In a few months Verilator will be completely incompatible with RH 6.  You are urged to [request to IT to] upgrade your OS immediately as running that old of an OS has serious security and other issues, (besides annoyances of bugs like these ;).

closed 2020-04-20 by wsnyder