verilator/verilator #2272: Parameter as 2D Array fails, not inferred as constant?

drichmond, 2020-04-20:
The following module compiles in VCS, but fails to compile in Verilator.

Verilator Error: 
```
%Error: example.sv:23:36: Expecting expression to be constant, but variable isn't const: 'hetero_type_vec_p'
                        : ... In instance foo
   23 |                  #(.hetero_type_p( hetero_type_vec_p[r-1][c] ))
      |                                    ^~~~~~~~~~~~~~~~~
%Error: example.sv:23:21: Can't convert defparam value to constant: Param 'hetero_type_p' of 'baz_i'
                        : ... In instance foo
   23 |                  #(.hetero_type_p( hetero_type_vec_p[r-1][c] ))
      |                     ^~~~~~~~~~~~~
%Error: Exiting due to 2 error(s)

```

```
module baz
  #(parameter integer hetero_type_p = -1)
   ();

endmodule

module foo
  #(
    parameter integer num_tiles_x_p = 4
    ,parameter integer num_tiles_y_p = 4
    //,parameter integer hetero_type_vec_p [0:num_tiles_y_p-2][0:num_tiles_x_p-1] = '{default:'{default:'0}}                                                                                                                                                                                                                  
    ,parameter integer hetero_type_vec_p [0:num_tiles_y_p-2][0:num_tiles_x_p-1] = '{num_tiles_y_p-1{'{num_tiles_x_p{'0}}}}
    )
   ();

   genvar r, c;

   for (r = 1; r < num_tiles_y_p; r = r+1)
     begin: y
        for (c = 0; c < num_tiles_x_p; c=c+1)
          begin : x
             baz
                 #(.hetero_type_p( hetero_type_vec_p[r-1][c] ))
                   baz_i
                   ();

          end
     end

endmodule
```

I can change `#(.hetero_type_p( hetero_type_vec_p[r-1][c] ))` to `#(.hetero_type_p( hetero_type_vec_p[0][0] ))` and the error does not change. 

Alternatively, I can pass hetero_type_vec_p as a parameter to baz and index it there. This fails with the error: 

```%Error: example.sv:10:40: Expecting expression to be constant, but variable isn't const: 'hetero_type_vec_p'
                        : ... In instance foo.y[1].x[0].baz_i
   10 |     ,parameter integer hetero_type_p = hetero_type_vec_p[idx_y][idx_x]
      |                                        ^~~~~~~~~~~~~~~~~
%Error: Exiting due to 1 error(s)
```

```
module baz
  #(
    parameter integer num_tiles_x_p = 4
    , parameter [$clog2(num_tiles_x_p)-1:0] idx_x  = 0
    , parameter integer num_tiles_y_p = 4
    , parameter [$clog2(num_tiles_y_p-1)-1:0] idx_y  = 0
    //,parameter integer hetero_type_vec_p [0:num_tiles_y_p-2][0:num_tiles_x_p-1] = '{default:'0}                                                                                                                                                                                                                             
    ,parameter integer hetero_type_vec_p [0:num_tiles_y_p-2][0:num_tiles_x_p-1] = '{num_tiles_y_p-1{'{num_tiles_x_p{'0}}}}

    ,parameter integer hetero_type_p = hetero_type_vec_p[idx_y][idx_x]
    )
   ();

endmodule

module foo
  #(
    parameter integer num_tiles_x_p = 4
    ,parameter integer num_tiles_y_p = 4
    //,parameter integer hetero_type_vec_p [0:num_tiles_y_p-2][0:num_tiles_x_p-1] = '{default:'0}                                                                                                                                                                                                                             
    ,parameter integer hetero_type_vec_p [0:num_tiles_y_p-2][0:num_tiles_x_p-1] = '{num_tiles_y_p-1{'{num_tiles_x_p{'0}}}}
    )
   ();

   genvar r, c;

   for (r = 1; r < num_tiles_y_p; r = r+1)
     begin: y
        for (c = 0; c < num_tiles_x_p; c=c+1)
          begin : x
             baz
                 #(.hetero_type_vec_p(hetero_type_vec_p))
                   baz_i
                   ();

          end
     end

endmodule
```

Any thoughts on how to fix this? Running with `Verilator 4.032 2020-04-04 rev v4.032` compiled on Centos 7. Happy to try solutions or hack a little bit of C++.

drichmond, 2020-04-20:
Update, this does work with 1D arrays:

```
module baz
  #(
    parameter integer num_tiles_y_p = 4
    , parameter [$clog2(num_tiles_y_p-1)-1:0] idx_y  = 0
    ,parameter integer hetero_type_vec_p [0:num_tiles_y_p-2]= '{num_tiles_y_p-1{'0}}

    ,parameter integer hetero_type_p = hetero_type_vec_p[idx_y]
    )
   ();

endmodule

module foo
  #(                                                                                                                                                                                                                                                                                                                          
    parameter integer num_tiles_x_p = 4                                                                                                                                                                                                                                                                                       
    ,parameter integer num_tiles_y_p = 4                                                                                                                                                                                                                                                                                      
    ,parameter integer hetero_type_vec_p [0:num_tiles_y_p-2] = '{num_tiles_y_p-1{'0}}                                                                                                                                                                                                                                         
    )
   ();

   genvar r, c;

   for (r = 1; r < num_tiles_y_p; r = r+1)
     begin: y
        baz
            #(.hetero_type_vec_p(hetero_type_vec_p))
        baz_i
            ();

     end

endmodule

```

wsnyder, 2020-04-21:
If you could look at this it would be great.  The first step is to make a test_regress format test (see internals.doc) that passes with another simulator (e.g. --vcs) and fails with verilator (--vlt or the default).

Then if you run with "--debug --debug-V3Simulate 9" you'll get obj_dir/tree files (described in internals.doc) and I suspect you'll see the internal simulator in src/V3Simulate.h which evaluates elaboration code is throwing an error.

Basically I suspect V3Simulate.h needs tweaks to handle arrays-of-arrays.

drichmond, 2020-04-21:
Yep, that's where it's failing. Want me to post the output?

drichmond, 2020-04-21:
Created a branch over in my fork: 
https://github.com/drichmond/verilator/pull/new/bug_twodim_parameter

drichmond, 2020-04-21:
Here's the failure log: 

[failure.txt](https://github.com/verilator/verilator/files/4507007/failure.txt)

wsnyder, 2020-04-21:
Yup, as suspected, let me know if you need help in the code.

drichmond, 2020-04-21:
That would be greatly appreciated -- I have no idea what I'm doing in that code. 

As a workaround though, I switched back to a 1D array. This is not on my critical path.

wsnyder, 2020-04-21:
Running with 

    v4make test_regress/t/t_EXAMPLE.pl --debug --dump-treei 9 --debugi-V3Param 9 --debugi-V3Simulate

We see this:

```
-cellparams:  PIN 0x5646a2e98780 <e195> {d30}  type_p -> VAR 0x5646a2e88000 <e10> {d9} @dt=0@  type_p GPARAM
-cellparams: 1: ARRAYSEL 0x5646a2ee3a80 <e3285#> {d30} @dt=0x5646a2de7500@(sw32)
-cellparams: 1:1: ARRAYSEL 0x5646a2ee39c0 <e3281#> {d30} @dt=0x5646a2dfe0d0@(sw32)u[0:3]
-cellparams: 1:1:1: VARREF 0x5646a2ed59e0 <e3252#> {d30} @dt=0x5646a2dfe1a0@(sw32)u[0:3]  twodim_p [RV] <- VAR 0x5646a2e88480 <e327#> {d19} u5=0x1 @dt=0x5646a2dfe\
1a0@(sw32)u[0:3]  twodim_p GPARAM
-cellparams: 1:1:2: CONST 0x5646a2f006e0 <e3272#> {d25} @dt=0x5646a2f042a0@(G/w2)  2'h0
-cellparams: 1:2: CONST 0x5646a2f00840 <e3305#> {d27} @dt=0x5646a2f042a0@(G/w2)  2'h1
%Error: t/t_EXAMPLE.v:30:16: Can't convert defparam value to constant: Param 'type_p' of 'baz_i'
```

So basically V3Simulate needs to handle an AstArraySel under an AstArraySel.  The code "handleArrayAssign" is a bit weedy and handles only 1D arrays and needs to be fixed to use recursion or otherwise handling 2+D as it should.

closed 2026-10-16 by wsnyder