verilator/verilator #2273: Build: Unable to install xml_py examples

imphil, 2020-04-21:
In a build on Ubuntu 16.04 (at least) I get the following warnings (the build still continues).

```
...
cd . \
; for p in examples/*/*.[chv]*  examples/*/Makefile* examples/*/CMakeLists.txt ; do \
  /usr/bin/install -c -m 644 $p /opt/buildcache/verilator/4.032/share/verilator/$p; \
done
/usr/bin/install: cannot create regular file '/opt/buildcache/verilator/4.032/share/verilator/examples/xml_py/sub.v': No such file or directory
/usr/bin/install: cannot create regular file '/opt/buildcache/verilator/4.032/share/verilator/examples/xml_py/top.v': No such file or directory
/usr/bin/install: cannot create regular file '/opt/buildcache/verilator/4.032/share/verilator/examples/xml_py/Makefile': No such file or directory
/bin/sh ./src/mkinstalldirs /opt/buildcache/verilator/4.032/share/pkgconfig
mkdir /opt/buildcache/verilator/4.032/share/pkgconfig
/usr/bin/install -c -m 644 verilator.pc /opt/buildcache/verilator/4.032/share/pkgconfig
/usr/bin/install -c -m 644 verilator-config.cmake /opt/buildcache/verilator/4.032/share/verilator
/usr/bin/install -c -m 644 verilator-config-version.cmake /opt/buildcache/verilator/4.032/share/verilator

Installed binaries to /opt/buildcache/verilator/4.032/bin/verilator
Installed man to /opt/buildcache/verilator/4.032/share/man/man1
Installed examples to /opt/buildcache/verilator/4.032/share/verilator/examples
...
```

It's probably a missing `-D` flag to install to create the subdirectories. I haven't debugged further, but I've seen it now a couple of times and thought I'd file this issue to not forget about it (or enable someone else to beat me to fixing it) :smile:

wsnyder, 2020-04-21:
Thanks for reporting it. As it happens this is a duplicate of #2251 which is fixed in git & the next release.

closed 2020-04-21 by wsnyder

imphil, 2020-04-21:
Argh I should have seen that. Sorry for the noise.