verilator/verilator #2275: Verilator does not create instances in a generate statement

srgatos6, 2020-04-22:
I am learning to use verilator. For this purpose I coded a parameterized gray count counter. It was written in two styles.

The first version is synthesizable RTL which used a verilog function to generate the toggle bits of the counter. This version simulated in verilog without an issue. The file containing this counter is the following:

src/rtl/gray_counter_FN.sv

The second version is still synthezible but is written as structural code. There are sub modules for the gray counter first bit, the intermediate bits and a last bit. These submodules are interconnected in an N-bit gray counter using the System Verilog generate statement.

When simulating this code Verilator outputs only zeroes. Further debugging using the -Wall option of Verilator showed that the clock, reset and count signals of the counter are not used which implies that the instances of the gray code bits are not created. This code simulated with iverilog and VCS without any issue but produces only zeroes on Verilator.

clock reset count q (structural) q (rtl)
                 100 1 1           0 0 000000 000000  
                 200 1 1           0 0 000000 000000  
                 300 1 0           0 0 000000 000000  
                 400 1 0           0 0 000000 000000  
                 500 1 0           0 0 000000 000000  
                 600 1 0           0 0 000000 000000  
                 700 1 0           0 0 000000 000000  
                 800 1 0           0 1 000000 000000  
                 900 1 0           1 1 000000 000001  
                1000 1 0           2 1 000000 000011  
                1100 1 0           3 1 000000 000010  
                1200 1 0           4 1 000000 000110  
                1300 1 0           5 1 000000 000111  
                1400 1 0           6 1 000000 000101  
                1500 1 0           7 1 000000 000100  
                1600 1 0           8 1 000000 001100 

You can see that the RTL code simulates correctly (it is counting in gray) but the structural code is stuck at zero.

The same code simulates correctly in iverilog (see that both counters count in gray).

clock reset count q 
                   0 1 1           0 0 xxxxxx xxxxxx  
                 100 1 0           0 0 000000 000000  
                 200 1 0           0 0 000000 000000  
                 300 1 0           0 0 000000 000000  
                 400 1 0           0 0 000000 000000  
                 500 1 0           0 0 000000 000000  
                 600 1 0           0 0 000000 000000  
                 700 1 0           0 0 000000 000000  
                 800 1 0           0 1 000000 000000  
                 900 1 0           1 1 000001 000001  
                1000 1 0           2 1 000011 000011  
                1100 1 0           3 1 000010 000010  
                1200 1 0           4 1 000110 000110  
                1300 1 0           5 1 000111 000111  
                1400 1 0           6 1 000101 000101  
                1500 1 0           7 1 000100 000100  
                1600 1 0           8 1 001100 001100  

The same code simulates correctly in iverilog.

Running verilator with the -Wall flags shows the problem. The instances are not generated and therefore clock, count and reset are not driving anything.

%Warning-UNUSED: src/rtl/gray_counter_wrapper.sv:20: Bits of signal are not used: q2[-1]
%Warning-UNUSED: src/rtl/gray_counter_wrapper.sv:21: Bits of signal are not used: q3[-1]
%Warning-UNUSED: src/rtl/gray_counter_wrapper.sv:22: Signal is not driven, nor used: q4
%Warning-UNUSED: src/rtl/gray_counter_wrapper.sv:23: Signal is not driven, nor used: q5
%Warning-UNUSED: src/rtl/gray_counter_N.sv:11: Signal is not used: clock
%Warning-UNUSED: src/rtl/gray_counter_N.sv:12: Signal is not used: reset
%Warning-UNUSED: src/rtl/gray_counter_N.sv:13: Signal is not used: count
%Warning-UNDRIVEN: src/rtl/gray_counter_N.sv:15: Signal is not driven: q
%Warning-UNUSED: src/rtl/gray_counter_N.sv:17: Bits of signal are not driven, nor used: zero[6:0]
%Warning-UNUSED: src/rtl/gray_counter_N.sv:17: Bits of signal are not used: zero[-1]
%Warning-UNUSED: src/rtl/gray_counter_N.sv:18: Bits of signal are not driven, nor used: toggle[6:0]
%Warning-UNUSED: src/rtl/gray_counter_N.sv:18: Bits of signal are not used: toggle[-1]
%Warning-UNUSED: src/rtl/gray_counter_FN.sv:17: Signal is not driven, nor used: zero
%Warning-UNUSED: src/rtl/gray_counter_FN.sv:18: Signal is not driven, nor used: toggle

Which explains why the structural code does not produce an output.

To reproduce the problem please untar the attached test case and do the following:

% make gray
% gray

Will run a verilator simulation which shows the problem

% make igray
% igray

Will run iverilog which works without a problem.

Thaks
Sergio

[gray.tar.gz](https://github.com/verilator/verilator/files/4519276/gray.tar.gz)

wsnyder, 2020-04-22:
Sorry you are having troubles.  I noticed you are running verilator differently from icarus - you have -GBITS=6.  When I remove that it works (but with 5 bits). The the problem is "-GBITS=6" is overriding your parameters at places other than your top level causing misconnects.

wsnyder, 2020-04-23:
P.S. Good writeup.

It's perhaps arguable if -G should override non-top levels like this.  If you can prove on another simulator that supports -G it ONLY overrides the top level, then we'll entertain fixing it - or better please provide a patch to fix it.  Thanks.

closed 2020-04-22 by wsnyder

srgatos6, 2020-04-23:
Hi Wilson
Thanks for your reply.
I tested it with -GBITS=5 same issue. You are right!!!. It is the GBITS which creates problems not the generate statement.

Running again:
verilator --top-module gray_counter_wrapper -GBITS=5 +1800-2017ext+sv  +1364-1995ext+v --cc src/rtl/gray_counter_wrapper.sv src/rtl/gray_counter_N.sv src/rtl/gray_counter_FN.sv src/rtl/gray_counter_cell.sv src/rtl/gray_counter_last_cell.sv -Wno-WIDTH -Wno-UNOPTFLAT --exe gray_counter_tb.cpp

clock reset count q 
                 100 1 1           0 0 00000 00000  
                 200 1 1           0 0 00000 00000  
                 300 1 0           0 0 00000 00000  
                 400 1 0           0 0 00000 00000  
                 500 1 0           0 0 00000 00000  
                 600 1 0           0 0 00000 00000  
                 700 1 0           0 0 00000 00000  
                 800 1 0           0 1 00000 00000  
                 900 1 0           1 1 00000 00001  
                1000 1 0           2 1 00000 00011  
                1100 1 0           3 1 00000 00010  
                1200 1 0           4 1 00000 00110  
                1300 1 0           5 1 00000 00111  
                1400 1 0           6 1 00000 00101  
                1500 1 0           7 1 00000 00100  
                1600 1 0           8 1 00000 01100  
                1700 1 0           9 1 00000 01101  
                1800 1 0          10 1 00000 01111     

Removing the GBITS=5 solves it as you mentioned in your email.

verilator --top-module gray_counter_wrapper  +1800-2017ext+sv  +1364-1995ext+v --cc src/rtl/gray_counter_wrapper.sv src/rtl/gray_counter_N.sv src/rtl/gray_counter_FN.sv src/rtl/gray_counter_cell.sv src/rtl/gray_counter_last_cell.sv -Wno-WIDTH -Wno-UNOPTFLAT --exe gray_counter_tb.cpp
clock reset count q 
                 100 1 1           0 0 00000 00000  
                 200 1 1           0 0 00000 00000  
                 300 1 0           0 0 00000 00000  
                 400 1 0           0 0 00000 00000  
                 500 1 0           0 0 00000 00000  
                 600 1 0           0 0 00000 00000  
                 700 1 0           0 0 00000 00000  
                 800 1 0           0 1 00000 00000  
                 900 1 0           1 1 00001 00001  
                1000 1 0           2 1 00011 00011  
                1100 1 0           3 1 00010 00010  
                1200 1 0           4 1 00110 00110  
                1300 1 0           5 1 00111 00111  
                1400 1 0           6 1 00101 00101  
                1500 1 0           7 1 00100 00100  
                1600 1 0           8 1 01100 01100  
                1700 1 0           9 1 01101 01101 

Editing the wrapper as follows works with 6 bits.
git diff src/rtl/gray_counter_wrapper.sv
diff --git a/src/testcases/gray/src/rtl/gray_counter_wrapper.sv b/src/testcases/gray/src/rtl/gray_counter_wrapper.sv
index 0f62e5b..01cc530 100644
--- a/src/testcases/gray/src/rtl/gray_counter_wrapper.sv
+++ b/src/testcases/gray/src/rtl/gray_counter_wrapper.sv
@@ -5,7 +5,7 @@ module gray_counter_wrapper (
     count
 );
 
-   parameter    BITS  = 5;
+   parameter    BITS  = 6;
    parameter    STATES = 1 << BITS;
    
    input        clock;

clock reset count q 
                 100 1 1           0 0 000000 000000  
                 200 1 1           0 0 000000 000000  
                 300 1 0           0 0 000000 000000  
                 400 1 0           0 0 000000 000000  
[... 26 lines cut ...]

clock reset count q 
                   0 1 1           0 0 xxxxxx xxxxxx    <-- Notice the x's which shows it is an iverilog run
                 100 1 0           0 0 000000 000000  
                 200 1 0           0 0 000000 000000  
                 300 1 0           0 0 000000 000000  
                 400 1 0           0 0 000000 000000  
                 500 1 0           0 0 000000 000000  
                 600 1 0           0 0 000000 000000  
                 700 1 0           0 0 000000 000000  
                 800 1 0           0 1 000000 000000  
                 900 1 0           1 1 000001 000001  
                1000 1 0           2 1 000011 000011  
                1100 1 0           3 1 000010 000010  
                1200 1 0           4 1 000110 000110  
                1300 1 0           5 1 000111 000111  
                1400 1 0           6 1 000101 000101  
                1500 1 0           7 1 000100 000100 

Maybe you can add support to it since I verilog does honor the top level parameter. I cannot test it now with VCS (got laid off so no access to a commercial tool) but I recall that it worked.

BTW. Excellent simulator!!! I am impressed (started using it yesterday). No frills much better than all that UVM stuff which obfuscates the testing. I love the fact that the test bench (wrapper) is synthesizable (sans the prints). That means that you could download a whole pass/fail test on an FPGA or emulator and test it at lightning speed which is impossible with UVM without having to write a lot of PLI code.
Great work!!!

RegardsSergio

srgatos6, 2020-04-23:
Hi Wilson

One more thing. I did an experiment with -GBITS=7.
As expected I got only zeroes on the structural code but the one that is defined with functions worked with seven bits without any code changes.
I think it is a bug. There seems to be an interaction between the generate statement and the -G flag which does not show on the synthesizable behavioral code. So the BITS parameter is passed to one model but not to the other.
See below.
RegardsSergio
verilator -GBITS=7 --top-module gray_counter_wrapper +1800-2017ext+sv  +1364-1995ext+v --cc src/rtl/gray_counter_wrapper.sv src/rtl/gray_counter.sv src/rtl/gray_counter_N.sv src/rtl/gray_counter_FN.sv src/rtl/gray_counter_no_neg.sv src/rtl/gray_counter_cell.sv src/rtl/gray_counter_last_cell.sv -Wno-WIDTH -Wno-UNOPTFLAT --exe gray_counter_tb.cpp
make -j -C obj_dir -f Vgray_counter_wrapper.mk Vgray_counter_wrapper

gray

clock reset count q 
                 100 1 1           0 0 00000 0000000 0000000 00000
                 200 1 1           0 0 00000 0000000 0000000 00000
                 300 1 0           0 0 00000 0000000 0000000 00000
                 400 1 0           0 0 00000 0000000 0000000 00000
                 500 1 0           0 0 00000 0000000 0000000 00000
                 600 1 0           0 0 00000 0000000 0000000 00000
                 700 1 0           0 0 00000 0000000 0000000 00000
                 800 1 0           0 1 00000 0000000 0000000 00000
                 900 1 0           1 1 00001 0000000 0000001 00001
                1000 1 0           2 1 00011 0000000 0000011 00011  <---- The two columns in bold are the relevant. The five bit columns are not parameterizable.                1100 1 0           3 1 00010 0000000 0000010 00010
                1200 1 0           4 1 00110 0000000 0000110 00110
                1300 1 0           5 1 00111 0000000 0000111 00111
                1400 1 0           6 1 00101 0000000 0000101 00101
                1500 1 0           7 1 00100 0000000 0000100 00100
                1600 1 0           8 1 01100 0000000 0001100 01100
                1700 1 0           9 1 01101 0000000 0001101 01101
                1800 1 0          10 1 01111 0000000 0001111 01111
                1900 1 0          11 1 01110 0000000 0001110 01110
                2000 1 0          12 1 01010 0000000 0001010 01010
                2100 1 0          13 1 01011 0000000 0001011 01011
                2200 1 0          14 1 01001 0000000 0001001 01001
                2300 1 0          15 1 01000 0000000 0001000 01000
                2400 1 0          16 1 11000 0000000 0011000 11000
                2500 1 0          17 1 11001 0000000 0011001 11001
                2600 1 0          18 1 11011 0000000 0011011 11011
                2700 1 0          19 1 11010 0000000 0011010 11010
                2800 1 0          20 1 11110 0000000 0011110 11110
                2900 1 0          21 1 11111 0000000 0011111 11111
                3000 1 0          22 1 11101 0000000 0011101 11101
                3100 1 0          23 1 11100 0000000 0011100 11100
                3200 1 0          24 1 10100 0000000 0010100 10100
                3300 1 0          25 1 10101 0000000 0010101 10101
                3400 1 0          26 1 10111 0000000 0010111 10111
                3500 1 0          27 1 10110 0000000 0010110 10110
                3600 1 0          28 1 10010 0000000 0010010 10010
                3700 1 0          29 1 10011 0000000 0010011 10011
                3800 1 0          30 1 10001 0000000 0010001 10001
                3900 1 0          31 1 10000 0000000 0010000 10000
                4000 1 0          32 1 00000 0000000 0110000 00000
                4100 1 0          33 1 00001 0000000 0110001 00001
                4200 1 0          34 1 00011 0000000 0110011 00011
                4300 1 0          35 1 00010 0000000 0110010 00010
                4400 1 0          36 1 00110 0000000 0110110 00110
                4500 1 0          37 1 00111 0000000 0110111 00111
                4600 1 0          38 1 00101 0000000 0110101 00101
                4700 1 0          39 1 00100 0000000 0110100 00100
    On Wednesday, April 22, 2020, 8:33:29 PM CDT, Sergio Ramirez <srgatos6@yahoo.com> wrote:  
 
  Hi Wilson
Thanks for your reply.
I tested it with -GBITS=5 same issue. You are right!!!. It is the GBITS which creates problems not the generate statement.

Running again:
verilator --top-module gray_counter_wrapper -GBITS=5 +1800-2017ext+sv  +1364-1995ext+v --cc src/rtl/gray_counter_wrapper.sv src/rtl/gray_counter_N.sv src/rtl/gray_counter_FN.sv src/rtl/gray_counter_cell.sv src/rtl/gray_counter_last_cell.sv -Wno-WIDTH -Wno-UNOPTFLAT --exe gray_counter_tb.cpp

clock reset count q 
                 100 1 1           0 0 00000 00000  
[... 87 lines cut ...]

clock reset count q 
                   0 1 1           0 0 xxxxxx xxxxxx    <-- Notice the x's which shows it is an iverilog run
                 100 1 0           0 0 000000 000000  
                 200 1 0           0 0 000000 000000  
                 300 1 0           0 0 000000 000000  
                 400 1 0           0 0 000000 000000  
                 500 1 0           0 0 000000 000000  
                 600 1 0           0 0 000000 000000  
                 700 1 0           0 0 000000 000000  
                 800 1 0           0 1 000000 000000  
                 900 1 0           1 1 000001 000001  
                1000 1 0           2 1 000011 000011  
[... cut ...]
Maybe you can add support to it since I verilog does honor the top level parameter. I cannot test it now with VCS (got laid off so no access to a commercial tool) but I recall that it worked.

BTW. Excellent simulator!!! I am impressed (started using it yesterday). No frills much better than all that UVM stuff which obfuscates the testing. I love the fact that the test bench (wrapper) is synthesizable (sans the prints). That means that you could download a whole pass/fail test on an FPGA or emulator and test it at lightning speed which is impossible with UVM without having to write a lot of PLI code.
Great work!!!

RegardsSergio

wsnyder, 2020-04-23:
icarus -D sets a define.  Maybe you meant -P.

Verilator seems to match VCS behavior, namely any submodule's parameter will get overridden. As only your structural model sets BITS to other than the top's value, it's the only one affected.

NC always wants a hierarchy.  Icarus only does top modules.  So, as Verilator is compatible with VCS and no one else agrees I'm going to keep -G as-is.

If you want to still use -G, simply change the name of BITS only in your top level's parameter, e.g.

    parameter TOP_BITS = 5;
    localparam BITS = TOP_BITS;

srgatos6, 2020-04-23:
OK ThanksSergio

srgatos6, 2020-04-23:
Hi Wilson:
I found a work around that is a good compromise. If I use -D to pass a flag I can override the quirk with the -G option. Here is what I did.
`ifndef TOP_BITS
 `define TOP_BITS = 5
`endif

module gray_counter_wrapper (
    clock,
    reset,
    count
);

   parameter    BITS  = `TOP_BITS;
   parameter    STATES = 1 << BITS;

I kept BITS as a parameter (not local param) to the wrapper and I can affect it by changing the value of TOP_BITS using -DTOP_BITS=7. The reason for keeping it as a parameter is that the wrapper can be instantiated by an iverilog test bench.

In the iverilog test bench (not used by verilator) I did the same so and then I pass BITS to the instance of the wrapper.
That way I get consistent results with iverilog and verilator.
Thanks a log for your help. And again thanks for releasing such a great simulator to the community.\
Cheers 
Sergio
7 bit simulation

iverilog -DTOP_BITS=7 -s gray_counter_wrapper_tb -g2012 -o igray src/rtl/gray_counter_wrapper.sv src/rtl/gray_counter.sv src/rtl/gray_counter_N.sv src/rtl/gray_counter_FN.sv src/rtl/gray_counter_no_neg.sv src/rtl/gray_counter_cell.sv src/rtl/gray_counter_last_cell.sv testbench/gray_counter_tb.sv testbench/gray_counter_wrapper_tb.sv

clock reset count q 
                   0 1 1           0 0 xxxxx xxxxxxx xxxxxxx xxxxx
                 100 1 0           0 0 00000 0000000 0000000 00000
                 200 1 0           0 0 00000 0000000 0000000 00000
                 300 1 0           0 0 00000 0000000 0000000 00000
                 400 1 0           0 0 00000 0000000 0000000 00000
                 500 1 0           0 0 00000 0000000 0000000 00000
                 600 1 0           0 0 00000 0000000 0000000 00000
                 700 1 0           0 0 00000 0000000 0000000 00000
                 800 1 0           0 1 00000 0000000 0000000 00000
                 900 1 0           1 1 00001 0000001 0000001 00001
                1000 1 0           2 1 00011 0000011 0000011 00011
                1100 1 0           3 1 00010 0000010 0000010 00010
                1200 1 0           4 1 00110 0000110 0000110 00110

verilator -DTOP_BITS=7 --top-module gray_counter_wrapper +1800-2017ext+sv  +1364                                                                                   -1995ext+v --cc src/rtl/gray_counter_wrapper.sv src/rtl/gray_counter.sv src/rtl/                                                                                   gray_counter_N.sv src/rtl/gray_counter_FN.sv src/rtl/gray_counter_no_neg.sv src/                                                                                   rtl/gray_counter_cell.sv src/rtl/gray_counter_last_cell.sv -Wno-WIDTH -Wno-UNOPT                                                                                   FLAT --exe gray_counter_tb.cpp  

clock reset count q 
                 100 1 1           0 0 00000 0000000 0000000 00000
                 200 1 1           0 0 00000 0000000 0000000 00000
                 300 1 0           0 0 00000 0000000 0000000 00000
                 400 1 0           0 0 00000 0000000 0000000 00000
                 500 1 0           0 0 00000 0000000 0000000 00000
                 600 1 0           0 0 00000 0000000 0000000 00000
                 700 1 0           0 0 00000 0000000 0000000 00000
                 800 1 0           0 1 00000 0000000 0000000 00000
                 900 1 0           1 1 00001 0000001 0000001 00001
                1000 1 0           2 1 00011 0000011 0000011 00011
                1100 1 0           3 1 00010 0000010 0000010 00010
                1200 1 0           4 1 00110 0000110 0000110 00110

    On Thursday, April 23, 2020, 7:38:36 AM CDT, Sergio Ramirez <srgatos6@yahoo.com> wrote:  
 
  OK ThanksSergio