verilator/verilator #2280: --build with CMake mitght not make sense

gezalore, 2020-04-23:
I am under the impression that the new `--build` with CMake does not make sense as the intention with CMake is to use the `verilate` function we provide here:
https://github.com/verilator/verilator/blob/6ed10b7fde1f2194c22345055f63294420b142d0/verilator-config.cmake.in#L116
from the user's CMake script, which adds rules to run verilator and include the objects in a target, e.g. like this:
https://github.com/verilator/verilator/blob/6ed10b7fde1f2194c22345055f63294420b142d0/examples/cmake_hello_c/CMakeLists.txt#L32-L38

When the user executes the CMake build, CMake will run Verilator with the `--make cmake`, slurp the emitted .cmake files which then tell the running build what files to pick up and add to the executable target, which is the ultimate goal of the CMake build..

I am struggling to make sense of how what we are doing right now could be useful with CMale:
https://github.com/verilator/verilator/blob/6ed10b7fde1f2194c22345055f63294420b142d0/src/Verilator.cpp#L590-L602
We just try to pick up the CMakeList.txt from the directory Verilator was invoked from? But surely if we have that CMakeList.txt, we would just use the `verilate` cmake function in that CMakeList.txt, otherwise.

The effect of this behaviour on t_

gezalore, 2020-04-23:
Furthermore, is it true that `--build` only makes sense when Verilator is invoked fromt he command line? In that case is there any point in trying to use CMake for `--build` (short of not having Make installed..)?

gezalore, 2020-04-23:
Also would add that I am not a CMake user so all the above might be wrong, I discovered this all trying to get t_flag_build_cmake to work on rerun

wsnyder, 2020-04-23:
Unfortunately I'm not really a cmake user either.  CC @yTakatsukasa @patstew 

The goal with this was to head towards #2206 which will do more than call the compiler (under make) - it might call other verilator runs, which in turn will have new files.  I guess we need to understand how that will fit in with cmake and that might help us understand how to settle this.

I also think there's value to have an option that makes Verilator look like other simulators - that is from the command line isolate the user from all build stuff.  If this was the only use model, and used only make that might be ok.

gezalore, 2020-04-23:
I agree the option to build is helpful, but if all the user wants is an exe, then as you said it's maybe ok to just leave it to Verilator to build it in whichever way it wants. As a user we wouldn't expect VCS or NC to build their models/internals with an external build system provided by us right?

patstew, 2020-04-23:
The cmake support should basically supplant the verilator executable so you don't run verilator directly at all. My gut reaction is that the bug title is correct, I'm not sure what the benefit of `verilator --build` as an alias for `cmake --build` is. As a way to automatically build with make on the other hand it looks handy.
From a brief look at #2206, it looks like support for that would need to be added to the cmake code, and I don't think that code would need this flag.

gezalore, 2020-04-24:
> The cmake support should basically supplant the verilator executable so you don't run verilator directly at all.
Just for the record, do you think then that `--make cmake` should really be a hidden option, used only by the cmake plugin? (Sorry if that's the wrong terminology)

patstew, 2020-04-24:
Perhaps? I don't necessarily see the need to actively hide it since the documentation and examples tell you how to go about using cmake so people aren't likely to use it by accident. I don't think people will ever need to use it, but in theory someone could want to use the file lists it spits out with their own scripts I suppose, I don't see why we should disable that.

yTakatsukasa, 2020-04-24:
I am sorry that I don't use verilator with cmake, but I think calling verilator directly and using cmake can bel useful in some cases.
`verilator -j --make cmake -MAKEFLAGS "-G Ninja" --build ... ` can be faster than `verilator -j --make gmake --build` if  `--output-split` is specified and hierarchical verilation is used.

Ninja is more efficient than GNU make if there are many sub directories.

 I agree that most people with use --build with --make gmake though.

gezalore, 2020-04-24:
> I think calling verilator directly and using cmake can bel useful in some cases.

I am sorry but I cannot see that yet. Let us ignore for this argument the speed of the build tool itself. In my experience large designs tend to be bound by the time it takes to run the C++ compiler and the speed of make is negligible even with hundreds of output files. We can also just teach Verilator to use the fastest build tool available when doing `--build`. With that in mind, can you think of cases when selecting the specific build system with `--make` when using `--build` would be useful?

I think the more fundamental question is whether `--build` and `--make` are mutually exclusive options, and when `--build` is used, Verilator is free to choose whatever method it thinks is best to build you the  executable?

yTakatsukasa, 2020-04-25:
For full build from scratch, build tool has little effect on the elapsed time, but when it comes to partial rebuild (only a few source file is updated), some tools are faster than gnu make. But let's leave this idea because it's a hypothesis on a hypothesis.

> We can also just teach Verilator to use the fastest build tool available when doing --build. With that in mind, can you think of cases when selecting the specific build system with --make when using --build would be useful? 

User may prefer specific build tool which is familiar to him/her regardless of slight build performance. For example, `-MAKEFLAGS` option is directly coupled with which build tools is used.

So your suggestion is to add a logic to tell error if both `--build` and `--make` are passed ?

I feel less benefit to explicitly disable a combination of options because the combination looks less popular.

Again I don't have strong opinion regarding cmake because I am not using it with verilator and have little knowledge of it.

gezalore, 2020-04-25:
> User may prefer specific build tool which is familiar to him/her regardless of slight build performance. For example, -MAKEFLAGS option is directly coupled with which build tools is used.

To me the `-MAKEFLAGS` seems to be there specifically to appease CMake (it does not seem to be needed to get any of the other tests work but t_flag_build_cmake.pl, which is the broken test that precipitated this issue). We already have `-CFLAGS`, `-LDFLAGS`  and you added `-j` which abstracts away a lot of the important things to control in a compile-then-link build. `-MAKEFAGS` can be useful to debug or tweak the build but I would certainly consider it an advanced option and the user shouldn't have to use it for simple things.

> So is your suggestion to add a logic to tell error if both --build and --make are passed ?
> 
> I feel less benefit to explicitly disable a combination of options because the combination looks less popular.

That is my suggestion. It is not motivated by concerns for pouplarity, but the desire to create a logically cohesive, easity to use and functional user interface. Verilator already has a higher barrier to entry than other simulators, `--build` is a great help in itself but it should not distract by providing alternatievs that don't (at least on the first order) affect the user experience.

Add to that, that currently `--build --make cmake` simply does not do someting that is useful, which violates the functionality. If we do decide to keep the option, then we will have to make it work, but I don't want to focus on that yet because I think the other aspects are more important to get right at this point.

I am thinking in terms of classes of users, of which I see 4 kinds:
- People integrating Verilator in their larger CMake or GNU Make based build system. They are well served with the pre-existing `--make cmake` and `--make gmake` options.
- People integrating Verilator in their larger build system using a different build tool. These could be helped by `--build` by getting the final exe/lib straight from invoking Verilator, at which point I doubt they care if internally Verilator used GNU Make, CMake, or even direcly invoked the toolchain [ignore build speed, we can always just do what is fastest], the exe/lib is there, the rest is Verilator's job.
- People using Verilator from the command line. These again are greatly helped by `--build` and `--exe`, but then again if all they want is an executable then the job to build it is delegated to Verilator.
- People who understand all moving parts of the system and can make do with anything they are provided, so they might appreciate simplicity, they are certainly not dependent on it.

It is possible there are people out there not covered by the above 4 groups, but without having some undersanding of their requirements it is hard to design tools for them. Once some of those requirements become clearer I would be more than happy to cater to them, but I would be cautious to add and subsequently support functionality that might or might not be useful without any further information, especially if at the same time that functionality detracts from the experience of the rest of the user base.

Don't get me wrong, I think `--build` is great, I just would like to make sure there is little opportunity for the user to spend brain cycles on decisions that are not important. Removing this cognitie load from the user enables them to focus on their end goal rather than having to think about whether they should use `--build --make gmake` or `--build --make cmake` when there is no appreciable difference, ignoring even the fact that the latter does not work right now.

Note also, I'm not even advocating for GNU Make in preference of CMake, I'm advocating for `--build` doing the right thing without further input, however Verilator happens to implement it.

> If --build --make cmake creates top CMakeLists.txt, would it be beneficial to Windows users where gmake is not necessarily installed ???

In my understanding, the Windows support is only wia Cygwin/MinGW, at which point gmake will be available.

yTakatsukasa, 2020-04-26:
Makes sense. I am fine with your suggestion.

Another approach would be providing one-stop-simpe-wrapper of the raw verilator, some commercial simulators offer such commend. But yes, maintaining such script is not free.

I respect your carefulness for usability.

gezalore, 2020-04-27:
Thanks all for the discussion. If everyone is happy then I propose doing the following to close this off:
- Issue an error when `--build` is used together `--make`
- When `--build` always use GNU Make for now to perform the build
- Update documentation & examples (also explain that -MAKEFLAGS should be rarely needed)
- Remove the broken t_flag_build_cmake test

Unless somebody feels strongly about implementing this I will try to do it myself sometime this week but ideally this should be in before the next release.

wsnyder, 2020-04-27:
Thanks for the great discussion - I agree with @gezalore's plan.

commit 849487da (2020-04-30): Modify --build to be a standalone option (#2294)

- Issue an error when --build is used together with --make
- When given --build, always use GNU Make to perform the build
- Update documentation (examples were good as they were)
- Remove the broken t_flag_build_cmake test

Fixes #2280
closed 2020-04-30 by gezalore