verilator/verilator #2286: public_flat_rw @(posedge clk) on SV interface

nndurj, 2020-04-25:
_What would you like added/supported?_

Support  `/*verilator public_flat_rw @(posedge clk) */` on SV interfaces

_Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)_

### HDL
```verilog
interface Bus();
    logic a;

    modport src (output a);
endinterface

module t(
    input i_clk
);

    Bus b1 ()  /*verilator public_flat_rw @(posedge i_clk) */;

endmodule
```

### Error
```
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module t --vpi --public-flat-rw --prefix Vtop -o t -LDFLAGS "-L `dirname /home/nndurj/sv_test/tests/build/libs/x86_64/libvpi.so` -lvpi -lgpi -lcocotb -lgpilog -lcocotbutils"  /home/nndurj/sv_test/tests/../hdl/test.v /home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages/cocotb/share/lib/verilator/verilator.cpp
%Error: /home/nndurj/sv_test/tests/../hdl/test.v:11:16: syntax error, unexpected /*verilator public_flat_rw*/, expecting ',' or ';'
   11 |     Bus b1 ()  /*verilator public_flat_rw*/ @(posedge i_clk) /**/ ;
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: Exiting due to 1 error(s)
```
May we assist you in trying to fix this yourself?

yes

wsnyder, 2020-04-25:
Thanks for the report/suggestion.  I don't think it would make sense on the interface like that, as interfaces can have many signals with different timing relationships/public status.

Please try it on the "logic a" in the interface and see if that does close enough to what you want.

nndurj, 2020-04-28:
@wsnyder You are right, if there is different timing then this requested feature does not make any sense.  Adding  `/*verilator public_flat_rw @(posedge i_clk) */` in the interface definition has its own problems, I guess its not worth it. For now I can find other workarounds. Thanks.

wsnyder, 2020-04-28:
Great!

closed 2020-04-28 by wsnyder