verilator/verilator #2288: Generated C++ gives compiler errors despite code passing lint wih -Wall

jshaker000, 2020-04-27:
Verilator 4.032 2020-04-04 rev v4.030-50-gd3797ade9

`clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.032/share/verilator/include -I/usr/local/Cellar/verilator/4.032/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wall -O3 --std=c++17   -c -o Vfft__ALLcls.o Vfft__ALLcls.cpp
eIn file included from Vfft__ALLcls.cpp:3:
./Vfft.cpp:1344:58: error: invalid operands to binary expression ('IData *' (aka 'unsigned int *') and 'unsigned int')
                                                         >> 7U) 
                                                         ^  ~~
./Vfft.cpp:1351:81: error: invalid operands to binary expression ('IData *' (aka 'unsigned int *') and 'unsigned int')
                                                                                >> 7U)))))))) 
                                                                                ^  ~~
./Vfft.cpp:1363:58: error: invalid operands to binary expression ('IData *' (aka 'unsigned int *') and 'unsigned int')
                                                         >> 7U) 
                                                         ^  ~~
./Vfft.cpp:1370:81: error: invalid operands to binary expression ('IData *' (aka 'unsigned int *') and 'unsigned int')
                                                                                >> 7U)))))))) 
                                                                                ^  ~~
4 errors generated.
make[1]: *** [Vfft__ALLcls.o] Error 1
make: *** [obj_dir/Vfft] Error 2
`

Attached is my project (which isnt the finest code, and I am still debugging it, but I woul expect it to compile).

My most recent change to to project was to instantiale inst_rndoI and inst_rndoQ in fft.v (the top level module). Without those two, the code works correctly.

I am wondering if my verilog is malformed and verilator did not catch it, or if the tool just had a hard time interpretting what I was trying to do?
[rtl.zip](https://github.com/verilator/verilator/files/4539704/rtl.zip)

wsnyder, 2020-04-27:
Thanks for the example. I updated git to make it clear what's wrong:

```
%Error: t/t_inst_misarray2_bad.v:10:17: Illegal input port connection 'i_data', mismatch between port which is not an array, and expression which is an array. (IEEE 1800-2017 7.6)
                                      : ... In instance t
   10 |                .i_data(fft_oQ[6:0])
      |                 ^~~~~~

```

closed 2020-04-27 by wsnyder

jshaker000, 2020-04-27:
You're right, and that was a mistake in my code, but it is helpful to print that as an error.

Thank you!