verilator/verilator #2291: Internal error when adding an else block after $feof

baltazarortiz, 2020-04-29:
I tried to build a module that reads a file and uses the value to drive outputs, and got the following error:

```
$ verilator broken.v --cc
%Error: Command Failed /usr/local/bin/verilator_bin broken.v --cc
%Error: Verilator internal fault, sorry.  Consider trying --debug --gdbbt
```

Running the debug command gives the following:
```
$ verilator broken.v --cc --debug
Starting Verilator 4.032 2020-04-04 rev v4.032-5-gbfcd779c
- V3File.cpp:235:        --check-times failed: different command line
- V3ParseImp.cpp:223: parseFile: broken
  Preprocessing broken.v
- V3PreShell.cpp:148:     Reading broken.v
- V3ParseImp.cpp:270: Lexing broken.v
- V3LinkCells.cpp:194:Link Module: MODULE 0x55b962690000 <e3#> {c1} u4=0x55b96265ae10  BrokenFileReader  L0
dot -Tpdf -o ~/a.pdf obj_dir/Vbroken_001_linkcells.dot
- V3LinkLevel.cpp:46: modSortByLevel()
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_002_cells.tree
- V3LinkDot.h:36:     linkDotPrimary:
- V3LinkJump.cpp:266: linkJump:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_007_link.tree
- V3Param.cpp:849:    param:
- V3LinkDot.h:41:     linkDotParamed:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_009_paramlink.tree
- V3Dead.cpp:438:     deadifyModules:
- V3Width.cpp:4945:   width:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_011_width.tree
- V3Width.cpp:4988:   widthCommit:
- V3Const.cpp:2692:   constifyAllLive:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_013_const.tree
- V3Undriven.cpp:458: undrivenAll:
- V3AssertPre.cpp:154:assertPreAll:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_014_assertpre.tree
- V3Assert.cpp:406:   assertAll:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_015_assert.tree
- V3LinkLevel.cpp:96: wrapTop:
- V3Const.cpp:2665:   constifyAllLint:
%Error: export VERILATOR_ROOT=
%Error: /usr/local/bin/verilator_bin_dbg broken.v --cc --debug
```

I simplified my module to the point that it doesn't even open the file or have any outputs, and I still get the same error until I comment out the final `else begin` block (whether there are things inside the block or not):

```
module BrokenFileReader (                                                       
    input                     clock,                                            
    input                     reset,                                            
    input                     in1                                               
);                                                                              
                                                                                
string filename;                                                                
integer               f      ; // var to see if file exists                     
                                                                                
always @(posedge clock) begin                                                   
        if (f != 0) begin                                                       
                if (!$feof(f)) begin                                            
                        $display("Doing stuff with file.");                     
                end                                                             
                // Commenting out these two lines fixes the fault               
                // The error still occurs if you put a $display or something       
                // else inside the block                                        
                else begin                                                      
                end                                                             
        end                                                                     
end                                                                             
                                                                                
endmodule    
[... 164 lines cut ...]
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_083_const_cpp.tree
- V3Dead.cpp:462:     deadifyAll:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_084_deadAll.tree
- V3Reloop.cpp:254:   reloopAll:
- V3Depth.cpp:174:    depthAll:
- V3Branch.cpp:134:   branchAll:
- V3Cast.cpp:177:     castAll:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_087_cast.tree
- V3CCtors.cpp:144:   cctorsAll:
- V3CUse.cpp:112:     cUseAll:
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_088_cuse.tree
- V3EmitCInlines.cpp:135:emitcInlines:
- V3EmitCSyms.cpp:877:emitcSyms:
- V3EmitC.cpp:3503:   emitcTrace:
- V3EmitC.cpp:3488:   emitc:
- V3EmitXml.cpp:352:  emitxml:
- V3StatsReport.cpp:237:statsReport:
- V3EmitMk.cpp:260:   emitmk:
- V3Os.cpp:83:        export VERILATOR_ROOT=/usr/local/share/verilator # Hardcoded at build time
- V3Os.cpp:83:        export SYSTEMC_ARCH=linux # From sysname 'linux'
- V3Ast.cpp:1077:     Dumping obj_dir/Vbroken_990_final.tree

```

`verilator --version` gives  `Verilator 4.032 2020-04-04 rev v4.032-5-gbfcd779c`. Let me know if other information would be useful for debugging!

wsnyder, 2020-04-29:
Thanks for your good bug report.  These are nice to debug n a way, as a crash generally makes it easy to fix, which it was. Anyhow there's an optimization with logical not this happened to hit - surprising wasn't found earlier.

Fixed in git towards 4.034.

closed 2020-04-29 by wsnyder

baltazarortiz, 2020-04-29:
No problem, thank you for the incredibly fast response and patch (and for rewording the title, I wasn't sure what exactly was causing the issue). I do want to note that the bug occurred for me even if the second block was not empty. I'll pull master and verify that the fix covers that case too!

baltazarortiz, 2020-04-29:
Confirmed that it fixes non-empty blocks as well!

Is master stable enough to use for now, or should I try to stick to releases? If it makes a difference, I'm working on researchy things, not production hardware designs.

wsnyder, 2020-04-29:
Master is usually stable enough for general use, it just is more likely to hit portability issues.

baltazarortiz, 2020-04-29:
Good to know, thanks so much!