verilator/verilator #2292: OS X/clang static member sizeof issue

horvathp, 2020-04-29:
Trying to start with verilator on OS X Catalina (10.15.4)
Apple clang version 11.0.3 (clang-1103.0.32.59)
Target: x86_64-apple-darwin19.4.0
(I'm not even sure if this is a supported combo)

make test fails (and every other example fails as well) because of this:
 
```
g++  -I.  -MMD -I/Users/hp/src/verilator/test_regress/../include -I/Users/hp/src/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC    -c -o verilated.o /Users/hp/src/verilator/test_regress/../include/verilated.cpp
/Users/hp/src/verilator/test_regress/../include/verilated.cpp:262:75: error: invalid use of member 'm_ser' in static member function
size_t Verilated::serialized2Size() VL_PURE { return sizeof(VerilatedImp::m_ser); }
                                                            ~~~~~~~~~~~~~~^~~~~
1 error generated.
```
I'm not sure why this 
https://github.com/verilator/verilator/blob/b44efe7ef781fc140ae3279d937695363ab6802d/include/verilated.cpp#L262
would be invalid. Quick googling reveals that there used to be a bug in clang
https://stackoverflow.com/questions/35901577/is-it-illegal-get-sizeof-non-static-member-of-struct-nested-within-class-templ
which is similar to this issue.

Is it possible to work this around somehow?
Thanks,
Peter

wsnyder, 2020-04-29:
Yes, seems a compiler bug.  Try using ...VerilatedImp::s_s.m_ser...  Adding the s_s may make it happy. If this works please consider making a pull request to fix it. Thanks.

horvathp, 2020-04-29:
That has definitely helped, but there are still problems:

```
g++  -I.  -MMD -I/Users/hp/src/verilator/include -I/Users/hp/src/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC    -c -o Vt_a1_first_cc__ALLslow.o Vt_a1_first_cc__ALLslow.cpp
ar -cr Vt_a1_first_cc__ALL.a Vt_a1_first_cc__ALLfast.o Vt_a1_first_cc__ALLslow.o
ranlib Vt_a1_first_cc__ALL.a
g++    Vt_a1_first_cc__main.o verilated.o verilated_vcd_c.o Vt_a1_first_cc__ALL.a    -o Vt_a1_first_cc -lm -lstdc++
ld: warning: ignoring file Vt_a1_first_cc__ALL.a, building for macOS-x86_64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture x86_64:
  "Vt_a1_first_cc::final()", referenced from:
      _main in Vt_a1_first_cc__main.o
  "Vt_a1_first_cc::trace(VerilatedVcdC*, int, int)", referenced from:
      _main in Vt_a1_first_cc__main.o
  "Vt_a1_first_cc::eval_step()", referenced from:
      Vt_a1_first_cc::eval() in Vt_a1_first_cc__main.o
  "Vt_a1_first_cc::Vt_a1_first_cc(char const*)", referenced from:
      _main in Vt_a1_first_cc__main.o
  "Vt_a1_first_cc::~Vt_a1_first_cc()", referenced from:
      _main in Vt_a1_first_cc__main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Vt_a1_first_cc] Error 1
```
Also I'm wondering why is the compiler trying to link against libstdc++ instead of libc++.

wsnyder, 2020-04-29:
Google the message and it suggests reinstalling https://apple.stackexchange.com/questions/373749/stack-install-fails-with-linker-errors-when-building-system-filepath-after-updat

horvathp, 2020-04-30:
My bad, wiping homebrew and everything else fixed the issue. Thanks for the tip.
Now it mostly works. The static library test fails, of course. (The stock OS X grep does not work correctly, as is often the case, but forcing the homebrew variant works fine.)

I have opened a very modest pull request to address the sizeof issue, as you suggested.

```
 -- VERILATE top module ----------------------------------------
 Use the SystemVerilog wrapper (verilated_secret.sv) and the
 library (libverilated_secret.a) generated from the previous
 step
---------------------------------------------------------------
/Users/hp/src/verilator/bin/verilator  -cc -Os -x-assign 0 -Wall --trace --exe -LDFLAGS '-L../obj_dir_secret -lverilated_secret -static' top.v obj_dir_secret/verilated_secret.sv sim_main.cpp

-- COMPILE entire design --------------------------------------
/Library/Developer/CommandLineTools/usr/bin/make -j 4 -C obj_dir -f Vtop.mk
g++  -I.  -MMD -I/Users/hp/src/verilator/include -I/Users/hp/src/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o sim_main.o ../sim_main.cpp
g++  -I.  -MMD -I/Users/hp/src/verilator/include -I/Users/hp/src/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o verilated.o /Users/hp/src/verilator/include/verilated.cpp
g++  -I.  -MMD -I/Users/hp/src/verilator/include -I/Users/hp/src/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o verilated_dpi.o /Users/hp/src/verilator/include/verilated_dpi.cpp
g++  -I.  -MMD -I/Users/hp/src/verilator/include -I/Users/hp/src/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o verilated_vcd_c.o /Users/hp/src/verilator/include/verilated_vcd_c.cpp
/usr/bin/perl /Users/hp/src/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop__Dpi.cpp Vtop__Trace.cpp > Vtop__ALLfast.cpp
/usr/bin/perl /Users/hp/src/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop__Syms.cpp Vtop__Trace__Slow.cpp > Vtop__ALLslow.cpp
g++  -I.  -MMD -I/Users/hp/src/verilator/include -I/Users/hp/src/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o Vtop__ALLfast.o Vtop__ALLfast.cpp
g++  -I.  -MMD -I/Users/hp/src/verilator/include -I/Users/hp/src/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o Vtop__ALLslow.o Vtop__ALLslow.cpp
ar -cr Vtop__ALL.a Vtop__ALLfast.o Vtop__ALLslow.o
ranlib Vtop__ALL.a
g++    sim_main.o verilated.o verilated_dpi.o verilated_vcd_c.o Vtop__ALL.a   -L../obj_dir_secret -lverilated_secret -static  -o Vtop -lm -lstdc++
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [Vtop] Error 1
make[1]: *** [run] Error 2
make: *** [examples] Error 10
```

wsnyder, 2020-04-30:
Glad that fixed it, thanks for the patch, pushed that.

closed 2020-04-30 by wsnyder