verilator/verilator #2293: Crash on dynamically sized array

teqdruid, 2020-04-29:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Not a full one by the RTL change was `byte unsigned DataOutBuffer[TYPE_SIZE_BYTES-1:0];` to  `   byte unsigned DataOutBuffer[];`

I wouldn't be surprised if the latter wasn't supported, but it's a crash rather than an error so I'm reporting it.

May we assist you in trying to fix this yourself? Help please

Stack trace:
```
warning: Error disabling address space randomization: Operation not permitted
No stack.
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Starting Verilator 4.032 2020-04-04 rev v4.032
- V3File.cpp:223:        --check-times failed: no input obj_dir/Vcosim_test__verFiles.dat
- V3ParseImp.cpp:223: parseFile: Cosim_DpiPkg
  Preprocessing Cosim_DpiPkg.sv
- V3PreShell.cpp:148:     Reading Cosim_DpiPkg.sv
- V3ParseImp.cpp:270: Lexing Cosim_DpiPkg.sv
- V3ParseImp.cpp:223: parseFile: Cosim_Endpoint
  Preprocessing Cosim_Endpoint.sv
- V3PreShell.cpp:148:     Reading Cosim_Endpoint.sv
- V3ParseImp.cpp:270: Lexing Cosim_Endpoint.sv
- V3ParseImp.cpp:223: parseFile: driver
  Preprocessing driver.sv
- V3PreShell.cpp:148:     Reading driver.sv
- V3ParseImp.cpp:270: Lexing driver.sv
- V3LinkCells.cpp:194:Link Module: PACKAGE 0x5613ed379b30 <e3#> {c14} u4=0x5613ed37ebc0  Cosim_DpiPkg  L0 [LIB]
- V3LinkCells.cpp:194:Link Module: PACKAGE 0x5613ed3923c0 <e124#> {a0} u4=0x5613ed3924f0  __024unit  L0 [LIB]
- V3LinkCells.cpp:194:Link Module: MODULE 0x5613ed392c80 <e126#> {d17} u4=0x5613ed392db0  Cosim_Endpoint  L0
- V3LinkCells.cpp:194:Link Module: MODULE 0x5613ed3c1480 <e1175#> {e3} u4=0x5613ed3c15b0  cosim_test  L0
- V3LinkCells.cpp:214:Link --top-module: MODULE 0x5613ed3c1480 <e1175#> {e3} u1=0x5613ed392b00 u4=0x5613ed3c15b0  cosim_test  L0
dot -Tpdf -o ~/a.pdf obj_dir/Vcosim_test_001_linkcells.dot
- V3LinkLevel.cpp:46: modSortByLevel()
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_002_cells.tree
- V3LinkDot.h:36:     linkDotPrimary: 
- V3LinkJump.cpp:266: linkJump: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_007_link.tree
- V3Param.cpp:849:    param: 
- V3LinkDot.h:41:     linkDotParamed: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_009_paramlink.tree
- V3Dead.cpp:438:     deadifyModules: 
- V3Width.cpp:4945:   width: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_011_width.tree
- V3Width.cpp:4988:   widthCommit: 
- V3Const.cpp:2692:   constifyAllLive: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_013_const.tree
- V3Undriven.cpp:458: undrivenAll: 
- V3AssertPre.cpp:154:assertPreAll: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_014_assertpre.tree
- V3Assert.cpp:406:   assertAll: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_015_assert.tree
- V3LinkLevel.cpp:96: wrapTop: 
- V3Const.cpp:2665:   constifyAllLint: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_017_const.tree
- V3SplitVar.cpp:1291:splitVariable: 
- V3Inst.cpp:590:     dearrayAll: 
- V3LinkDot.h:46:     linkDotArrayed: 
- V3Begin.cpp:296:    debeginAll: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_022_begin.tree
- V3Tristate.cpp:1388:tristateAll: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_023_tristate.tree
- V3Unknown.cpp:475:  unknownAll: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_024_unknown.tree
- V3Inline.cpp:716:   inlineAll: 
- V3Ast.cpp:1077:     Dumping obj_dir/Vcosim_test_025_inline.tree
[... 75 lines cut ...]
#50 0x00005613eb640239 in AstNVisitor::iterateChildren (this=0x7fffce1f2190, nodep=0x5613ed450930) at ../V3Ast.h:2482
#51 0x00005613eba9341c in TaskVisitor::visit (this=0x7fffce1f2190, nodep=0x5613ed450930) at ../V3Task.cpp:1136
#52 0x00005613eb63b436 in AstNVisitor::visit (this=0x7fffce1f2190, nodep=0x5613ed450930) at ./V3Ast__gen_visitor.h:177
#53 0x00005613eb8e84e8 in AstModule::accept (this=0x5613ed450930, v=...) at ../V3AstNodes.h:2029
#54 0x00005613eb65e1c5 in AstNode::iterateAndNext (this=0x5613ed450930, v=...) at ../V3Ast.cpp:804
#55 0x00005613eb65de68 in AstNode::iterateChildren (this=0x5613ed368f40, v=...) at ../V3Ast.cpp:766
#56 0x00005613eb640239 in AstNVisitor::iterateChildren (this=0x7fffce1f2190, nodep=0x5613ed368f40) at ../V3Ast.h:2482
#57 0x00005613eba946d1 in TaskVisitor::visit (this=0x7fffce1f2190, nodep=0x5613ed368f40) at ../V3Task.cpp:1289
#58 0x00005613eb63b672 in AstNVisitor::visit (this=0x7fffce1f2190, nodep=0x5613ed368f40) at ./V3Ast__gen_visitor.h:188
#59 0x00005613eb845e26 in AstNetlist::accept (this=0x5613ed368f40, v=...) at ../V3AstNodes.h:7115
#60 0x00005613eb640212 in AstNVisitor::iterate (this=0x7fffce1f2190, nodep=0x5613ed368f40) at ../V3Ast.h:2479
#61 0x00005613eba94945 in TaskVisitor::TaskVisitor (this=0x7fffce1f2190, nodep=0x5613ed368f40, statep=0x7fffce1f20d0) at ../V3Task.cpp:1302
#62 0x00005613eba88167 in V3Task::taskAll (nodep=0x5613ed368f40) at ../V3Task.cpp:1528
#63 0x00005613eb631c7c in process () at ../Verilator.cpp:246
#64 0x00005613eb632c77 in main (argc=12, argv=0x7fffce1f2548, env=0x7fffce1f25b0) at ../Verilator.cpp:586
A debugging session is active.

	Inferior 1 [process 546] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]
%Error: export VERILATOR_ROOT=
%Error: gdb /usr/local/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --cc Cosim_DpiPkg.sv Cosim_Endpoint.sv driver.sv --top-module cosim_test --exe driver.cpp /esi/linbin/src/core/cosim/cosim_dpi_server/libEsiCosimDpiServer.so --debug --gdbbt" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed gdb /usr/local/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --cc Cosim_DpiPkg.sv Cosim_Endpoint.sv driver.sv --top-module cosim_test --exe driver.cpp /esi/linbin/src/core/cosim/cosim_dpi_server/libEsiCosimDpiServer.so --debug --gdbbt" -ex 'set width 0' -ex 'bt' -ex 'quit'

```

wsnyder, 2020-04-29:
Thanks for reporting this; dynamic arrays are supposed to be supported.

Looks like this is an argument to a DPI/public or other function, can you please post the function - ideally see if you can get a crash with code that just declares and calls that function.

teqdruid, 2020-04-29:
Here's the signature:

```
import "DPI-C" sv2c_cosimserver_ep_tryget = 
  function int cosim_ep_tryget(
    input  int unsigned endpoint_id,
    output byte unsigned data[],
    inout  int unsigned size_bytes
    );
```

And the usage:
```
rc = cosim_ep_tryget(ENDPOINT_ID, DataOutBuffer, data_limit);
```

Is this what you're looking for? I can provide all the RTL on the side... it's going to be OSS at some point, but has yet to be released.

I'm actually also having a DPI issue and in trying to figure that out I hit this.

teqdruid, 2020-04-29:
Oh and I forgot: This is Verilator v4.032

wsnyder, 2020-04-29:
Ok, that's a DPI open array which is a bit different than an associative array.  @gezalore as you were recently in there would you mind taking a look if you have a chance?

wsnyder, 2020-04-29:
@teqdruid please try git master, the improvements of Geza's aren't released yet and might fix it, there were a ton of fixes.

teqdruid, 2020-04-29:
With master Verilator doesn't crash, but when I try to build the output:

```
verilator --cc --top-module cosim_test Cosim_DpiPkg.sv Cosim_Endpoint.sv driver.sv --exe driver.cpp /esi/linbin/src/core/cosim/cosim_dpi_server/libEsiCosimDpiServer.so
make -j -C obj_dir -f Vcosim_test.mk Vcosim_test
make[1]: Entering directory '/esi/src/core/cosim/rtl/obj_dir'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o Vcosim_test__ALLfast.o Vcosim_test__ALLfast.cpp
In file included from /usr/include/c++/7/ext/alloc_traits.h:36:0,
                 from /usr/include/c++/7/bits/stl_construct.h:61,
                 from /usr/include/c++/7/deque:62,
                 from /usr/local/share/verilator/include/verilated_heavy.h:29,
                 from Vcosim_test.h:11,
                 from Vcosim_test.cpp:5,
                 from Vcosim_test__ALLfast.cpp:3:
/usr/include/c++/7/bits/alloc_traits.h: In instantiation of 'struct std::allocator_traits<std::allocator<unsigned char&> >':
/usr/include/c++/7/ext/alloc_traits.h:50:10:   required from 'struct __gnu_cxx::__alloc_traits<std::allocator<unsigned char&> >'
/usr/include/c++/7/bits/stl_deque.h:464:21:   required from 'class std::_Deque_base<unsigned char&, std::allocator<unsigned char&> >'
/usr/include/c++/7/bits/stl_deque.h:831:11:   required from 'class std::deque<unsigned char&, std::allocator<unsigned char&> >'
/usr/local/share/verilator/include/verilated_heavy.h:262:44:   required from 'class VlQueue<unsigned char&>'
Vcosim_test.cpp:129:252:   required from here
/usr/include/c++/7/bits/alloc_traits.h:392:27: error: forming pointer to reference type 'unsigned char&'
       using pointer = _Tp*;
                           ^
/usr/include/c++/7/bits/alloc_traits.h:395:39: error: forming pointer to reference type 'unsigned char&'
       using const_pointer = const _Tp*;
                                       ^
In file included from /usr/include/c++/7/bits/stl_construct.h:61:0,
                 from /usr/include/c++/7/deque:62,
                 from /usr/local/share/verilator/include/verilated_heavy.h:29,
                 from Vcosim_test.h:11,
                 from Vcosim_test.cpp:5,
                 from Vcosim_test__ALLfast.cpp:3:
/usr/include/c++/7/ext/alloc_traits.h: In instantiation of 'struct __gnu_cxx::__alloc_traits<std::allocator<unsigned char&> >':
/usr/include/c++/7/bits/stl_deque.h:464:21:   required from 'class std::_Deque_base<unsigned char&, std::allocator<unsigned char&> >'
/usr/include/c++/7/bits/stl_deque.h:831:11:   required from 'class std::deque<unsigned char&, std::allocator<unsigned char&> >'
/usr/local/share/verilator/include/verilated_heavy.h:262:44:   required from 'class VlQueue<unsigned char&>'
Vcosim_test.cpp:129:252:   required from here
/usr/include/c++/7/ext/alloc_traits.h:66:23: error: no members matching '__gnu_cxx::__alloc_traits<std::allocator<unsigned char&> >::_Base_type {aka std::allocator_traits<std::allocator<unsigned char&> >}::allocate' in '__gnu_cxx::__alloc_traits<std::allocator<unsigned char&> >::_Base_type {aka struct std::allocator_traits<std::allocator<unsigned char&> >}'
     using _Base_type::allocate;
                       ^~~~~~~~
/usr/include/c++/7/ext/alloc_traits.h:67:23: error: no members matching '__gnu_cxx::__alloc_traits<std::allocator<unsigned char&> >::_Base_type {aka std::allocator_traits<std::allocator<unsigned char&> >}::deallocate' in '__gnu_cxx::__alloc_traits<std::allocator<unsigned char&> >::_Base_type {aka struct std::allocator_traits<std::allocator<unsigned char&> >}'
     using _Base_type::deallocate;
                       ^~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/c++/7/bits/c++allocator.h:33:0,
                 from /usr/include/c++/7/bits/allocator.h:46,
                 from /usr/include/c++/7/deque:61,
                 from /usr/local/share/verilator/include/verilated_heavy.h:29,
                 from Vcosim_test.h:11,
                 from Vcosim_test.cpp:5,
                 from Vcosim_test__ALLfast.cpp:3:
/usr/include/c++/7/ext/new_allocator.h: In instantiation of 'class __gnu_cxx::new_allocator<unsigned char&>':
/usr/include/c++/7/bits/allocator.h:108:11:   required from 'class std::allocator<unsigned char&>'
/usr/include/c++/7/bits/stl_deque.h:549:14:   required from 'struct std::_Deque_base<unsigned char&, std::allocator<unsigned char&> >::_Deque_impl'
/usr/include/c++/7/bits/stl_deque.h:633:19:   required from 'class std::_Deque_base<unsigned char&, std::allocator<unsigned char&> >'
/usr/include/c++/7/bits/stl_deque.h:831:11:   required from 'class std::deque<unsigned char&, std::allocator<unsigned char&> >'
/usr/local/share/verilator/include/verilated_heavy.h:262:44:   required from 'class VlQueue<unsigned char&>'
Vcosim_test.cpp:129:252:   required from here
/usr/include/c++/7/ext/new_allocator.h:63:26: error: forming pointer to reference type 'unsigned char&'
       typedef _Tp*       pointer;
                          ^~~~~~~
/usr/include/c++/7/ext/new_allocator.h:64:26: error: forming pointer to reference type 'unsigned char&'
       typedef const _Tp* const_pointer;
                          ^~~~~~~~~~~~~
In file included from /usr/include/c++/7/deque:61:0,
[... cut ...]
     void __Vdpiimwrap_cosim_ep_tryget__Vdpioc2_TOP__Cosim_DpiPkg(const IData/*31:0*/ endpoint_id, VlQueue<CData/*7:0*/ &>  data, IData/*31:0*/(&  size_bytes), IData/*31:0*/(&  cosim_ep_tryget__Vfuncrtn));
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vcosim_test_Cosim_DpiPkg.h:37:10: note:   no known conversion for argument 2 from 'VlQueue<unsigned char>' to 'VlQueue<unsigned char&>'
/usr/local/share/verilator/include/verilated.mk:211: recipe for target 'Vcosim_test__ALLfast.o' failed
make[1]: *** [Vcosim_test__ALLfast.o] Error 1
make[1]: Leaving directory '/esi/src/core/cosim/rtl/obj_dir'
```

gezalore, 2020-04-29:
I am happy to take a look at this.

gezalore, 2020-05-01:
I am still looking at this, but just as an FYI, the last sentence of Section 7.8 of IEEE 1800-2017 says:

> A dynamic array or queue shall not be passed as an actual argument if the
DPI formal argument has unsized dimensions and an output direction mode.

Which applies to the code in this issue, so we need an error. I presume the restriction is due to the inability to handle memory management on the C size, though I'm less falmiliar around witht this part of the DPI. Also, it looks to me that input/inout do not work either which probably should.

wsnyder, 2020-05-01:
I believe this is an open array (35.5.6.1) aka unsized array, not an associative array.  These have limited handling in Verilator but are (well were intended to be before this bug) handled in Verilator. See also the t_dpi_open.v test.

gezalore, 2020-05-01:
What I meant, is that the import statement from above specifies the _formal_ argument as an open array:
```
import "DPI-C" sv2c_cosimserver_ep_tryget = 
  function int cosim_ep_tryget(
    input  int unsigned endpoint_id,
    output byte unsigned data[],
    inout  int unsigned size_bytes
    );
```

And what triggers the bug is changing the type of the _actual_ argument passed to the 2nd formal in the above from bland unpacked array to a dynamic array, as per the top of this issue:

> Not a full one by the RTL change was byte unsigned DataOutBuffer[TYPE_SIZE_BYTES-1:0]; to byte unsigned DataOutBuffer[];

Problem 1: This actually is illegal due to 7.7 I guess this is because output arguments have an undefined initial state (see 35.5.1.2), and there is no mechanism/API for changing the size of an array from the C side. It is arguable whether it should or should not work if you have already allocated the dynamic array, but that point in 7.7 suggest it need not to meet the standard. I think Verilator could make them work if we wanted to (check big 3 support).

Problem 2; The open array _formal_ argument should be able to bind to either a bland unpacked array, dynamic array or queue of a compatible type (subject to not violating 7.7 I guess), but dynamic array/qeueue are currently broken in Verilator. The bug causing the compile error above actually is here, putting the `&` for referce argument declaration in the wrong place (inside the template parameter):
https://github.com/verilator/verilator/blob/6e9008fb5a06396ea44e5410822b6d3e810137c9/src/V3AstNodes.cpp#L335-L345
I am reviewing the whole open array handling right now to see if there are other issues, there is at least one more suspisious one here (upper limit should be `udims()`):
https://github.com/verilator/verilator/blob/6e9008fb5a06396ea44e5410822b6d3e810137c9/include/verilated.cpp#L2445-L2449

gezalore, 2020-05-08:
Turns out we are currently missing support for passing dynamic arrays to DPI open array arguments. I raised #2315 to track this missing feature. #2316 adds an error for the above case, so I will be closing this once that is merged.

closed 2020-05-08 by gezalore