verilator/verilator #2298: Change to C++17 and libc++?

kj6msg, 2020-04-30:
I have installed Verilator on macOS via Homebrew and it's fantastic. However, verilate uses an older C++ standard and does not use libc++ on the Mac. For me to compile my test benches with C++17 and libc++, I have to compile and link outside of verilate, which seems to defeat the purpose. Any tips?

wsnyder, 2020-04-30:
You mean configure adds the wrong compiler flags for your system?  What are the CFG_* lines at top of include/verilated.mk?

kj6msg, 2020-04-30:
> You mean configure adds the wrong compiler flags for your system? What are the CFG_* lines at top of include/verilated.mk?

Thanks for that... I clearly wasn't looking hard enough. Digging in there, looks like it "hard codes" the libstd++ standard library, vice libc++ which would be appropriate for macOS. Fixed.

wsnyder, 2020-04-30:
So it works when the LIBS += is removed?    Not sure if we can generally remove that, might need a configure rule.

kj6msg, 2020-04-30:
I changed it to -stdlib=libc++ on my own installation. I could just remove that and clang++ would still select the correct stdlib.

Interestingly enough I started messing with CMake and Verilator after posting this and there is no problem there.

wsnyder, 2020-04-30:
I committed removing the LIBS.  I think this code was a leftover from very old requirements; I tried several systems and they all seemed ok, so we'll hope for the best.

closed 2020-04-30 by wsnyder