verilator/verilator #2299: Array revives gated clock

marcowidmer, 2020-04-30:
Here is a minimal example:

```verilog
module t(/*AUTOARG*/
    // Inputs
    clk
);
    input clk;

    reg clk_en = 1'b0;
    wire clk_gated = clk & clk_en;
    wire [1:0] clks = {1'b0, clk_gated};

    always @(posedge clks[0])
        $display("ERROR: clks[0] should not be active!");
endmodule
```

In Verilator 4.032, this displays the error message at every tick of `clk`.

In the waveforms, I can see that clk_gated and clks[1:0] are correctly set to 0. However, the always block still triggers.

I had some limited success using `/* verilator var_split */` on `clks`. However, this does not work if I want to pass `clks` to a submodule.

marcowidmer, 2020-04-30:
This seems to be a bug in `GateClkDecompGraphVisitor` in V3Gate: It replaces the clock vector item (`clks[0]`) with the root clocker (`clk`) instead of the direct parent (`clk_gated`). This is wrong because `clk_gated` is derived from, but not equal to `clk`.

wsnyder, 2020-04-30:
Note also to prevent races in more general code you need verilator clock_enable attributes.

marcowidmer, 2020-04-30:
You are correct, the wire fixes this simple case. But that is because it gets optimized before V3Gate.

I am confident that this is not related to how the zero is produced. I can reproduce this in a lot of different ways, as long as the 0-generation is non-trivial enough that it is not optimized out.

```verilog
    reg clk_en;
    initial
        clk_en = 1'b0;
```
or
```verilog
module zero (
    output wire z
);
    assign z = 0;
endmodule

// ...
    wire clk_en;
    zero i0(.z(clk_en));
```
cause the same error.

I am also aware of `/* verilator clock_enable */`, but this does not fix anything in this case.

wsnyder, 2020-04-30:
The decomposition in general wasn't properly stopping on anything unexpected.

Fixed in git towards 4.034.

closed 2020-04-30 by wsnyder

marcowidmer, 2020-05-04:
I can confirm that this fixes the issue. Thank you for the very fast response!