verilator/verilator #2300: Issue in AND-reduction

jackkoenig, 2020-04-30:
I have created a repo that illustrates the issue: https://github.com/jackkoenig/verilator-and-reduction-bug
Commit of the above repo at time of filing issue: 9a38b944bd9909d32611ed3c0a920a3f49170fd5

Pasted from README:

It seems there is a bug in handling of AND-reduction (and possibly other reductions, I haven't checked)
for wires with >64-bits.
I have confirmed the buggy behavior in versions >= 4.026.

This test case compares the results of equivalent operations AND-reduce and checking equality with all ones for a 68-bit register:
```verilog
  assign and_reduce = &r;
  assign check_equal = r == 68'hffff_ffff_ffff_ffff_f;
```

To reproduce, setup verilator and run:
```
make run
```

On versions < 4.026 you will see the correct output:
```
reset = 0, r = 00000000000000000, and_reduce = 0, check_equal = 0
reset = 0, r = ffffffffffffffffe, and_reduce = 0, check_equal = 0
reset = 0, r = fffffffffffffffff, and_reduce = 1, check_equal = 1
```

For versions >= 4.026, you will see the incorrect output:
```
reset = 0, r = 00000000000000000, and_reduce = 1, check_equal = 0
reset = 0, r = ffffffffffffffffe, and_reduce = 0, check_equal = 0
reset = 0, r = fffffffffffffffff, and_reduce = 0, check_equal = 1
```

wsnyder, 2020-04-30:
Thanks for the good test.  This should have been covered in the test suite, sorry for the hole.

Fixed towards v4.034.

closed 2020-04-30 by wsnyder

albert-magyar, 2020-04-30:
Thanks! I have to say, the tree dumps with `--debug` make it really easy to track issues like this -- you can just watch the `REDAND` go away after expand!

johningalls-sifive, 2020-05-05:
Did this also effect bitwise-or/xor/xnor reductions?

wsnyder, 2020-05-05:
No, I tested the others too and didn't find any issue.