verilator/verilator #2303: Tristate does not work with multiple assignments

marcowidmer, 2020-05-04:
This creates correct tristate logic:
```verilog
always_comb begin
   if (condition)
      s = 1'b1;
   else
      s = 1'bz;
end
```

This however creates some sort of latch (the 1 is written to the signal, but never cleared):
```verilog
always_comb begin
   s = 1'bz;
   if (condition)
      s = 1'b1;
end
```

I am not sure if this is supposed to be supported, but the documentation does not mention anything.

Here is a testcase with two drivers that consecutively write to a tristate signal:
```verilog
`define BROKEN

module t(/*AUTOARG*/
    // Inputs
    clk
);
   input clk;

   int cyc = 0;
   always @(posedge clk) begin
      cyc <= cyc + 1;
      if (cyc == 99) begin
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

   // tristate signal
   reg s;

   // drive during cyc == 0
`ifdef BROKEN
   always_comb begin
      s = 1'bz;
      if (cyc == 0)
         s = 1'b1;
   end
`else
   always_comb begin
      if (cyc == 0)
         s = 1'b1;
      else
         s = 1'bz;
   end
`endif

   // drive during cyc == 1
   always_comb
      s = (cyc == 1) ? 1'b0 : 1'bz;

   // verify signal
   always @(posedge clk) begin
      if (cyc == 0 && s != 1'b1) begin
         $display("cyc 0: s = %d, expected 1!\n", s);
         $stop;
      end
      if (cyc == 1 && s != 1'b0) begin
         $display("cyc 1: s = %d, expected 0!\n", s);
         $stop;
      end
   end
endmodule
```
On Verilator 4.034, this fails with `cyc 1: s = 1, expected 0!`.

wsnyder, 2020-05-05:
Looked at this a bit.  Basically the current tristate resolution logic assumes that all tristates are generated in wire/assign statements.  Either this needs to be fixed - I fixed one problem there but another remains with how the enables are currently generated as wires, or an unsupported error should be thrown.

You got lucky with the workaround you tried in that Verilator was able to optimize it to something that then was supported

    assign s = (cyc == 0) ? 1'b1 : 1'bz;

AbacusIsMad, 2024-08-16:
In the latest versions (5.024, 5.026), I am able to reproduce this issue even with the first example (so both are incorrect now), with or without optimization:
``` systemverilog
always_comb begin
   if (condition)
      s = 1'b1;
   else
      s = 1'bz;
end
```
like the above, the `1` is written and never cleared. Luckily, the workaround of using wire/assign still works. Just wondering, does verilator plan to support tristates more fully in the future? Thanks

wsnyder, 2024-08-18:
There hasn't been any recent work on tristates.  If this is important to improve for you please consider working on a pull request.